jbruce12000/kiln-controller #184: Down ramp sensor temp gets stuck

MeltTechAu, 2024-07-22:
Upon entering a down ramp sensor temp stayed same temp got stuck at 805 degrees c it went into a hold or something and did not down ramp heater turned off though I had to restart pi for it to read temp again
Catch up off
I also have errors ignored except thermocouple break
Any ideas jbruce

jbruce12000, 2024-07-22:
not sure why a TC would get stuck at a temperature. A post of the logs would tell me more.

MeltTechAu, 2024-07-23:
Rerun same program and it worked

MeltTechAu, 2024-07-26:
Got stuck again after it went choppy went full power at 580 then did that for 20 mins up and down 25 degrees  c then thermo got stuck again on an upramp I’ll change out the board jbruce

MeltTechAu, 2024-07-26:
Have u got any beta ware you need testing

rondoc, 2024-07-26:
Sounds like hardware problem. Wiring? connections?

jbruce12000, 2024-08-05:
Was this a hardware problem? Can I close?

MeltTechAu, 2024-08-07:
ive tried another brand new unit does the same seems to be software to me i rerun another progrm 3 times and on the 3rd it got stuck on an upramp 13 minutes from the end of the run i put my old software in and hardware seems fine i have run 20 times on that. to me it seems like a bug i tried 3 different max 31856 chip from different batches in each unit hmmmm i also noticed if range error is set in config to true noconnection no longer works. currently im sticking to the old software with max 31855 seems very reliable ,ill keep trying your new software but this time try it with th max 31855 see if i get the same problem. i run these controllers daily it seems to be random where it gets stuck on the graph the line will just go horizontel timer still runs. i only tested on a fresh dl from you. 
really think its software issue jbruce as hardware seems fine only difference is the max 31856 chip it may only do it with that as you know i sell these units so hoping for reliability
i dont know why it would stick ive tried adding stuck sensor reinitialize to the program it didnt work. still got stuck ,only turning off  and on the unit it then resets the temp. im using harware spi 8 9 10 11.
 
oh i did add Adaptive tuning to your new software can i send you it so you can add it to your new software it seems to work to me but gets stuck the same as your new software main . still hoping you can fix i run this little kiln constantly testing have you got an email i can send the files as txt to you as i havnt upgraded github yet.
thanks for the hardwork you do mate apriciate it bigtime

jbruce12000, 2024-08-07:
I absolutely need to see the logs when this failure happens in order to troubleshoot the cause.

jbruce12000, 2024-08-07:
I also need a copy of config.py used when it gets stuck.

MeltTechAu, 2024-08-07:
how do i do that? never had to look at the logs yet? lol

here is some code i just got ai to make fixes in the old code that im still using for profile order and it works did it tonight.

i do 2 jobs but will focus on testing your new software ASAP once i know how to get the logs i can then give them to you
in kiln controller.py overwrite the old this is in numerical order

@app.get('/api/profiles')
def get_profiles():
    def extract_number(filename):
        import re
        match = re.search(r'\d+', filename)
        return int(match.group()) if match else float('inf')

    profile_files = sorted(
        [f for f in os.listdir(profile_path) if os.path.isfile(os.path.join(profile_path, f))],
        key=extract_number
    )
    profiles = []
    for filename in profile_files:
        with open(os.path.join(profile_path, filename), 'r') as f:
            profiles.append(json.load(f))
    return json.dumps(profiles)

also without using a grounding wire to the pi and the positive of the thermocoupe with the max 31855 1300 degs c is 1282 degs c tested using my work calibrator with grounding wire its within 1 deg C otherwise from 1000 degs onwards its out even on the new software havent tested it with calibrator on max 31856 just a thought ive done heaps of testing happy to supply you my fixes and add ons like adaptive tuning mate i want the new software to work so i can use othe thermocouple types i did also notice that when selecting s or r type max31856 that it starts in the minuses ambient temp like -15 deg c should be reading like 20 degs c but using the calibrator it sorts its self out after 100 deg c hmm i do more testing ive only sold 8 of these so far as i said i work at a kiln company through the day and do my controller stuff at night busy as but never to busy to make this software no 1 happy to help
cheers brett

MeltTechAu, 2024-08-07:
i tried zip loags /var/log/daemon.log does not exist and i get empty files hmmm i am using bookworm os is it different daemon log does not exist i tried logging a short run getting files with zero bytes if i zip log it as per instructions has it changed with os update let me know would love to send you logs can easily post a screen shot of the graph of the front end ill start testing when i know how to get the logs.

just figured out i can send you files through here below is Adaptive Tuning version give it a go i havent had time to tune it up right yet with correct pid settings. it does have stuck temp fix that did not work in it tried reinitalize sensor
and the code i sent earlier fixes numerical order issue for firing profiles in the new one just tested it
[adaptive sanitty cheecks.zip](https://github.com/user-attachments/files/16531378/adaptive.sanitty.cheecks.zip)
Here to help and be helped
enjoy playing with Adaptive Tuning but first lets sort this problem out
bed time for me 3 hrs sleep before work

MeltTechAu, 2024-08-07:
daemon log does not exist

let me knw how to get the logs i start testing

here is adaptive tuning i was messing with seems to work just need correct pid settings to work from i think just another option
download below
[adaptive sanitty cheecks.zip](https://github.com/user-attachments/files/16531572/adaptive.sanitty.cheecks.zip)
enjoy kp ki kd on state page will show it works also in terminal

i also did AI tuning but it would not save the output file for the AI regresion to read maybe you could look at that just another option make cool hey
[AI Adaptive Tunng.zip](https://github.com/user-attachments/files/16531594/AI.Adaptive.Tunng.zip)
 
bed time i gota go to my day job

MeltTechAu, 2024-08-08:
[image]
Got 
[image]
Got
to 84 degs c then went horizontally see graphs

MeltTechAu, 2024-08-08:
Hmm

MeltTechAu, 2024-08-08:
if you can fix this issue  above that would be great but can you also incorporate my adaptive tuning i made  into the fixed version please as standard and also the profile ordering here is a copy of your main i modded with a choice of adptive if needed can u make it standard in main.
config in below is what i been using less the addons  above graphs are from your fresh dl main kiln controller

adaptive tuning means one less thing people have to do full working program below but freezes like yours still

[adaptive Tuning.zip](https://github.com/user-attachments/files/16541051/adaptive.Tuning.zip)

i do have AI PID tuning im still looking into dam thing wont save output file for the regresion technique(i tinker around with that let me know when you have time to look at it if i cannot make it work)

ill try tommorrow with just the max31855 on your software didnt go to good with the max31856 didnt even make the first 1hr before freezing have to just keep running it by the graph it looks like the relay got  stuck on then froze that was a little different then before.

PS you can see adaptive tuning working in state file and also terminal kp ki kd adjusting hope you can fix the freezing issue mate please and thank you
just another feature to make it better
please try the adaptive tuning it just works
cheeers brett

MeltTechAu, 2024-08-08:
[adaptive Tuning.zip](https://github.com/user-attachments/files/16541389/adaptive.Tuning.zip)
incorporte that into it when fixed please

MeltTechAu, 2024-08-08:
New add on
 that works for you to test your state page with Adaptive Tuning for you to play with
[adaptive sanitty cheecks.zip](https://github.com/user-attachments/files/16541687/adaptive.sanitty.cheecks.zip)
download on pc linux seems to mangle it

rondoc, 2024-08-08:
@meltechau..Notin context but...Just notice you mentioned relay frozen..
are you using.mechanical.relays? The SS switches work so.much also the zero
crossing control reduces RFI that causes problems.

jbruce12000, 2024-08-08:
post the output of 

`ls -la /var/logs
`
also, try running ziplogs and post the output here. the won't give me everything, but some is better than none.

rondoc, 2024-08-08:
@MeltTechAu. I have had problems with connection to the pi. The female
connectors lose their spring very easily. Then you can get intermittent
problems.
I have considered soldering to the actual.pi pins but so far managed by
replacing the connectors. The atmosphere has a bearing on this and if the
pi is near the kiln then it might be a cause due to vapors of salts in the
air enough to corrode the pin inteface.
Ronald.

MeltTechAu, 2024-08-09:
JB i can not get ziplogs to work says permision denied also there is no daemon log hmmmmm i am using bookworm fresh install too only has kiln controller on it any  ideas i can post the front end 
 graphs and state page graphs etc tried using sudo no avail
 Rondoc i have reeewired the controllers incase of bad wiring everthing is firm and works flawlessly with JB old software
ill keep testing controller i run twice with the max31855 i put in seems fine so far maybe an issue oly when using the max3856 we shall figure it out im running the kiln 24 hours a day im using the same firing profiles as the failed max31856
can anyone confirm this issue thoses of you runnng the max31856 are you having similar issue
i tried moding the software to reinitialize thermocouple sensor to no avail i make it reboot instead as a reboot fixes stuck sensor.
yes i use 2 relays in series not SS as they stay cooler
 and its no PWM
Sorry JB about the logs 
cheers Brett

rondoc, 2024-08-09:
I have been using the 55 chip board for two years now with little problems.
It is ideal for K type thermocouples. Why change to the 56?

jbruce12000, 2024-08-10:
ok, if you are on bookworm, then you're using systemd by default. this means that all logs are handled by journalctl. try this...

`journalctl --since "2024-08-10 00:00:00" >mylog.txt
`

then take a look at the file you produced:

`ls -la mylog.txt
`

then maybe

`more mylog.txt
`

finally, gzip it and post it here...

`gzip mylog.txt
`

You need to use a time period that covers the failure for me to troubleshoot the problem. Get the right window with start and end date and times if you need...

`journalctl --since "2024-08-01" --until "2015-08-03"`

jbruce12000, 2024-08-11:
I think (and I cannot verify without the logs) that you're experiencing an OVUV fault on the max-31856. Here is how the spec sheet describes the error...

```
When the absolute input voltage at T+ or T- is negative or
greater than VDD, the Under-/Overvoltage Fault bit, Bit 1,
is set in the Fault Status register (0Fh) and the FAULT pin
asserts if not masked. Conversions are suspended while
the OVUV fault is present and will resume when the fault
is removed.
```

If this is the error you're getting, it will show up as a "voltage" exception in the logs. If you're ignoring exceptions... which you say you are, it would look like this...

`Problem reading temp (ignored) voltage`

jbruce12000, 2024-08-11:
If this is an OVUV problem, the ONLY way to fix it is to Clear Faults on the max-31856. There is NO provision to do this in adafruit's code today. This would be possible to add, but they would almost certainly reject the request to add it to their code via pull request.

This would leave you with... fix the hardware issue.

MeltTechAu, 2024-08-12:
well that didnt work

MeltTechAu, 2024-08-12:
[image]

[image]
Log didn’t show me up to that time I was heating it with a lighter it’s freezing temp on that error I dunno what’s going on mate

MeltTechAu, 2024-08-12:
Im sure im not the only one going to have this problem  this is sending me nuts does it on a fresh dl from here and my modded one adaptive tuning one as well I’m thinking I’ll have to make it reboot as that’s the only way to unstick temp can anyone else try it seems to work with max31855 ok but I really want type R. I don’t know wot I’m doing wrong

jbruce12000, 2024-08-12:
The logs you've shown in screenshots from the terminal prove this is a OVUV fault. If you want to continue to use the 31856, you must fix the hardware problem. Most likely you're getting noise voltage spikes.

Try these things:

- put a small capacitor across the two thermocouple wires (so in parallel).
- try twisting the thermocouple wires around each other
- try shortening the thermocouple wires
- try using shielded cable to connect to the thermocouple

@rondoc and others have more experience than I do with noise reduction. They will have suggestions too.

In the future, since you're running kiln-controller.py from the command line, to save the logs, do this:

`./kiln-controller.py 2>&1 |tee mylog.txt
`

jbruce12000, 2024-08-12:
@rondoc he has 31855 and runs it fine. He wants to run the 31856 because it supports R thermocouples.

His problem is thermocouple noise. He's seeing voltage spikes above the limit the 31856 allows. This causes the 31856 to stop sampling until the fault is masked and reset (neither of which the adafruit library supports today).

So, if anyone else has had trouble with noise, let MeltTechau know how you fixed it.

rondoc, 2024-08-12:
He should get zero crossing solid state switches. Relays make a lot of
electrical noise. The other noise suppression ideas you listed are good
also. Noise suppression is hard that is why i say get rid of the relays.
Its ok to use a relay as a safety cut off.

MeltTechAu, 2024-08-12:
[AI Adaptive Tunng.zip](https://github.com/user-attachments/files/16591776/AI.Adaptive.Tunng.zip)
heres something for you to play with jbruce i can not get it to save the output file for it to train the pid
would be good to add this if it worked just another option you may be able to get it working
even though the pid is pretty good once tuned ill try above suggestions thanks for the advice as always appreciated thank you all ..  the max 31855 is solid but as i said calibrator at work shows 1300 degs c ,kiln controller is 1282 deg c when you ground it out in my software it stablises the floating issue and reads within 1 deg c your new software doesnt handle grounding can you make it handle it? it reads 0 deg c if grounded

rondoc, 2024-08-12:
the chip 31855 doesnot allow for either of the inputs to be grounded. .

MeltTechAu, 2024-08-14:
i use negative on the pi to the positive of the thermocouple line, it  stops temp floating and makes it accurate to 1 deg c with the old software modded it works well. i modded the software to handle it.

jbruce12000, 2024-08-14:
Closing this. It's a hardware issue. There are no changes for me to make.

Changes to the adafruit 31856 library are possible that would allow for a reset after an OVUV error.

closed 2024-08-14 by jbruce12000