jbruce12000/kiln-controller #185: emergency shut off bug

jbruce12000, 2024-07-23:
if the emergency shutoff temp is exceeded. the kiln shuts down and if automatic restarts is set, it does... and should not.

rondoc, 2024-07-23:
Is this a problem? May be if emergency temp is exceeded  and alarm should
be set and require a manual overide if possible. It seems that this
exceeding temp. condition MUST require manual intevention

jbruce12000, 2024-07-23:
I think it's a bug. I have not tested the _main_ branch to see if it has the bug yet. soon.

captaincaden, 2024-07-23:
For what it's worth @jbruce12000, I'm currently all the way up to date (including additions from today), and when simulating a kiln to exceed the emergency shut off temp, it does NOT automatic restart for me. Behaves as expected. Automatic restarts and emergency shutoff both enabled.

jbruce12000, 2024-08-05:
Thanks for testing @captaincaden! Closing.

closed 2024-08-05 by jbruce12000