jbruce12000/kiln-controller #187: Add elapsed time to dashboard

solnus, 2024-08-21:
While the expected duration being displayed is nice, it would be helpful to display a current elapsed time as well.

rondoc, 2024-08-21:
1. I agree showing actual.local.time would be a "nice" feature .
2. What is status on stopping and resuming a run from where it stopped. I
know this has been discussed. If it is available can i add it without
changing my current system.?

jbruce12000, 2024-08-21:
Pause and Resume was added to the api. Someone requested this feature with power issues. He would call the api to pause at the current temp and use a generator until power came back where he'd resume. See the [api docs](https://github.com/jbruce12000/kiln-controller/blob/main/docs/api.md).

jbruce12000, 2024-08-21:
@solnus seems logical, but it's not so simple. if it's just displaying elapsed time since the start of the run, that's pretty easy. Is that what you want?

solnus, 2024-08-21:
@jbruce12000 yes, this feature request from my perspective is simply elapsed time since start of the run. Additional considerations (like pause/resume) aren't in the scope I had intended.