jbruce12000/kiln-controller #189: Add exit handler/watchdog for safety

solnus, 2024-08-21:
Currently if the process exits while the relay is closed, the relay remains closed. This could be a safety issue. I see a couple initial options, but I'm sure there are others:

- Add an exit handler that attempts to force the output pin to off state, or power off and hope the pin goes to hi-Z and shuts off the relay
- Add a linux watchdog for the process, such that if the process stops kicking the watchdog a safety mechanism kicks in

jbruce12000, 2024-08-21:
it is impossible to fully overcome stuck hardware. when SSRs fail, it's almost always a short circuit. this means that full power is applied until power is removed (even if the rpi is powered off).  There are other hw solutions, which you are free to explore, but I don't think they should be part of this codebase.

some folks have relays that trip if the kiln draws 100% etc.

solnus, 2024-08-21:
@jbruce12000 this request was more about software exceptions, not SSR hardware failure. If the process exits normally, or throws an exception and aborts (bugs are always possible), it should set the digital output pin to OFF or Open. I think the first step here could just be catching exceptions in main and disabling the pin when caught.