jbruce12000/kiln-controller #190: Install/run service from user-agnostic directory

solnus, 2024-08-22:
When trying to install the service using the provided script, I noticed that it failed due to assumptions about the username and path to the scripts. This should ideally not be necessary to hard-code. Here are a couple of ideas and I can implement whichever one is deemed best if there's some agreement.

Option 1:
- start-on-boot script installs a copy of the current scripts into /opt somewhere, and uses that path. Downside is that it requires updating /opt whenever there are changes, if development isn't being done in /opt.

Option 2:
- start-on-boot script creates a symlink for kiln-controller in /usr/local/bin (or /opt), so it still sources from the user directory, but the script installation is agnostic of the actual path

Option 3:
- Dynamically generate the init script to directy point at the kiln-controller in the current directory as of running start-on-boot

jbruce12000, 2024-08-22:
I really have no opinion on this. I'm willing to accept a PR for this. PR should be limited in scope and solve just THIS problem.

potus31, 2024-09-14:
Which files do you have to re-hardcode with /home/<HOSTNAME>/kiln-controller/

solnus, 2024-09-14:
The following files have hard-coded paths with the assumed pi username:

lib/init/kiln-controller.service
start-on-boot

wasexton, 2024-11-05:
Thanks!  I didn't name mine pi so this helped me get these changed so I could start on boot.

closed 2026-10-10 by jbruce12000