jbruce12000/kiln-controller #194: Error messages when testing with Thermocouple.py.

Thomas-Wendt, 2024-09-01:
When I test the TC with thermocople.py, the temperature values are unstable and I get the error messages ‘short circuit to ground’ and ‘short circuit to power’ between the measured values.

I have already exchanged the TC and the max 31855 and the power supply - without success.  

The setup of the kiln-controller went through without any error messages.

What else I noticed, during the boot from the RPI the LED from the SSR lights up in 1 sec. rhythm . Is that normal ?

Use a R-pi 4B with 7’ touch display. 

Any idea what could be wrong ? 

[image]

jbruce12000, 2024-09-03:
The reported values should NOT vary that much. When temps are static, the temps I read on my 31855 don't vary much. Certainly less than 1 degree F.

When you are booting up the PI, your SSR is switched on and off every second? I have never seen this happen before.

Have you double checked all the wiring to make sure you're connecting everything to the right pins? It seems like there is something fundamentally wrong.

jbruce12000, 2024-09-03:
Try twisting the leads of the thermocouple.
Try putting a small cap across the thermocouple inputs.
Make the thermocouple leads as short as you can.
Make sure ground on your pi is connected to something that is really grounded. In the US, our power outlets all have a ground. My PIs ground is connected to that.

solnus, 2024-09-04:
I agree with jbruce, this seems like the pins are not connected/configured properly. If something else is using the pin you're trying to use, you will see all sorts of odd behavior (even running kiln-controller service at the same time as thermocouple test will cause problems, make sure kiln-controller is not running when you're testing the thermocouple).

rondoc, 2024-09-04:
You need to separate problems.
1. Check the tc with a tc meter.
2. Check that the leads to the tc are just going to the tc. They MUST not
connect or even have high resistance to ground.
3. There must be no SSW signals until you in a run , firing.
4. As others mentioned re check wiring...

rondoc, 2024-09-04:
As an aside do not put power to kiln until uou have it all working. You can
do a trial run without the main power and should see the drive to the SSW
going on.

closed 2026-10-10 by jbruce12000