jbruce12000/kiln-controller #195: juggling of profiles fix numerical order ;) not issue any more

MeltTechAu, 2024-09-07:
juggling of profiles fix numerical order put in kilncontroller..py  :)
label like this  1186 C5 Bisc Std.json

@app.get('/api/profiles')
def get_profiles():
    def extract_number(filename):
        import re
        match = re.search(r'\d+', filename)
        return int(match.group()) if match else float('inf')

    profile_files = sorted(
        [f for f in os.listdir(profile_path) if os.path.isfile(os.path.join(profile_path, f))],
        key=extract_number
    )
    profiles = []
    for filename in profile_files:
        with open(os.path.join(profile_path, filename), 'r') as f:
            profiles.append(json.load(f))
    return json.dumps(profiles)

jbruce12000, 2024-09-08:
I think you're asking... Hey Jason, I fixed the ordering of profiles and I'd like you to merge this change with main... can you do that for me?

is that what you want?

MeltTechAu, 2024-09-08:
Hi Jason
Yes mate only if you want too! ;)
Have a Great Day!

mymatejohn, 2025-02-06:
Hi Jason and co, thank you for this excellent project, it truly is amazing. I have just installed and am testing your kiln controller so it is up to date, but my firing profiles are in no particular order in the dropdown list. I've labelled them according to MeltTechAu's comment above but I'm not sure if this adjustment has been implented. Do you have any suggestions?

[image]

gadgetmies, 2025-08-05:
Made this change in https://github.com/jbruce12000/kiln-controller/pull/238

closed 2026-10-10 by jbruce12000