jbruce12000/kiln-controller #196: Units missing from Kiln profile files

behanw, 2024-09-10:
The temperatures in a profile have no recorded unit, and are assumed to be in the unit which is in the configuration (temp_scale). However, if existing profiles written in Fahrenheit are read when temp_scale is set to Celcius, then things can go very wrong.

I propose that "unit" is added to the profile json and used to translate temperatures, when the profile json is read, into "temp_scale" if they differ.

I've written a Python script that can already do that conversion and adds "unit" if missing. I assume that if there is no unit, the temps are in Fahrenheit since the profiles currently in git are in Fahrenheit.

I propose changes to the Profile Class which will similarly convert the units appropriately for profiles being read, and then further changes which will write the unit to a saved profile.

behanw, 2024-09-10:
Oh I see part of this is already here. New profiles have "temp_units" added. However older profiles without this set are assumed to be in C, when they are clearly in F.

jbruce12000, 2024-10-06:
I wrote conversion code to convert to C every time any profile is saved. Try saving your old profile and if the conversion is not correct, re-open this issue.

closed 2024-10-06 by jbruce12000