jbruce12000/kiln-controller #197: MAX31865 Support (RTD for more accuracy at lower temps)

TheAdeo, 2024-11-06:
Hi, I am wondering how difficult it would be to add support for RTD amplifiers like the common MAX31865.
This is because for lower temperature applications like PID controlled convection ovens RTD offer much greater accuracy and repeatability and stability.
I know that these are not the applications for which this controller was written, but they would benefit greatly from being supported by it.

Thanks

jbruce12000, 2024-11-10:
Short answer... use a max-31856 which supports R type thermocouples without code changes.

The max-31865 probably would work fine but might require lots of code changes. I have not tried it. It would require uncommenting the module in requirements.txt and then closely examining the adafruit module to see how it handles errors. last, changes would need to be made to oven.py... TempSensorReal and ThermocoupleError classes would need to be created.

jbruce12000, 2024-11-10:
Probably a few hours of work plus hardware time.

I don't have a max-31865 or rtd thermocouples, so I've not tried to do this. I've certainly thought about it, but for my application[s] that type of accuracy/repeatability is not needed. I could see this being used by scientists.

jbruce12000, 2024-11-12:
Tell ya what, to make a small impact in the scientific community,  you buy the hw... I'll write and test the code.

I think the total for a MAX31865 plus an RTD is about $26 + tax.

https://www.adafruit.com/product/3328

TheAdeo, 2024-11-12:
Yeah I already have the MAX31865 and both a PT100 and some PT1000s, so I would be able to test it. Thank you very much <3

TheAdeo, 2024-11-12:
I don't want to scope creep you too much, but for a scientific precision application it would be better if you could set the controller operating frequency, as 2s intervals are too long for small and fast systems. And also I can't seem to find a way to measure and set the temperature in °C instead of °F

jbruce12000, 2024-11-12:
A couple of things... 

1. sensor_time_wait was tested at 1/10th of a second and seems to work, but this is likely to cause problems with clients
2. it's pretty unlikely that you have a system that heats and cools at a rate that requires at sensor_time_wait under 1s. Maybe, if it was like a high powered laser striking a plate in a vacuum of a small volume, but no oven that uses elements is ever gonna heat fast enough to need to take a sample every second.

Then again, I'm no scientist. Maybe you have a unique case.

TheAdeo, 2024-11-12:
Oh yeah if they're just settings in the config.py then it's good.
As for the sensor_time_wait, a shorter interval coupled to a RTD can lead to a much better accuracy and lower fluctuations, but yes, 1s is enough.

For the MAX31865: I can't spare that much money, but if you settle for [Aliexpress](https://it.aliexpress.com/item/1005004561372911.html) you can get the module plus a pt100 for just 5$, and I would be happy to pay that