jbruce12000/kiln-controller #199: New Build - Thermo Couple Issue

wasexton, 2024-11-14:
I am creating a new build and at first I had pins wired incorrectly and was receiving an error when running test-thermocouple but now I am getting 32.00F regardless of whether the thermo couple is connected to the 31855 or not.

(venv) admin@kiln:~/kiln-controller $ ./test-thermocouple.py
Software SPI selected for reading thermocouple
SPI configured as:

    config.spi_sclk = 17 BCM pin
    config.spi_mosi = 10 BCM pin
    config.spi_miso = 27 BCM pin
    config.spi_cs   = 22 BCM pin

board: RASPBERRY_PI_4B
thermocouple: adafruit max31855
Degrees displayed in f

2024-11-14 13:46:38.678077 32.00F
2024-11-14 13:46:39.678651 32.00F
2024-11-14 13:46:40.679211 32.00F
2024-11-14 13:46:41.679748 32.00F

I know I probably have something wrong but can not locate what it would be and after having validated all the power voltages and checked/rechecked wiring to ensure the correct pins, I do not know how to debug this.  Any assistance would be greatly appreciated. Thank you in advance for your time and consideration.

wasexton, 2024-11-14:
If this helps, here are the results of gpioreadall

+-----+------------+------+---+  Pi 4B   +---+------+-----------+-----+
| BCM |    Name    | Mode | V |  Board   | V | Mode | Name      | BCM |
+-----+------------+------+---+----++----+---+------+-----------+-----+
|        3.3v                 |  1 ||  2 |             5v             |
|   2 | INPUT      |      | 1 |  3 ||  4 |             5v             |
|   3 | INPUT      |      | 1 |  5 ||  6 |             GND            |
|   4 | INPUT      |      | 1 |  7 ||  8 | 1 |      | INPUT     |  14 |
|        GND                  |  9 || 10 | 1 |      | INPUT     |  15 |
|  17 | OUTPUT     |      | 0 | 11 || 12 | 0 |      | INPUT     |  18 |
|  27 | INPUT      |      | 1 | 13 || 14 |             GND            |
|  22 | OUTPUT     |      | 1 | 15 || 16 | 0 |      | OUTPUT    |  23 |
|        3.3v                 | 17 || 18 | 0 |      | INPUT     |  24 |
|  10 | OUTPUT     |      | 0 | 19 || 20 |             GND            |
|   9 | SPI0_MISO  |      | 0 | 21 || 22 | 0 |      | INPUT     |  25 |
|  11 | SPI0_SCLK  |      | 0 | 23 || 24 | 1 |      | OUTPUT    |   8 |
|        GND                  | 25 || 26 | 1 |      | OUTPUT    |   7 |
|   0 | INPUT      |      | 1 | 27 || 28 | 1 |      | INPUT     |   1 |
|   5 | INPUT      |      | 1 | 29 || 30 |             GND            |
|   6 | INPUT      |      | 1 | 31 || 32 | 0 |      | INPUT     |  12 |
|  13 | INPUT      |      | 0 | 33 || 34 |             GND            |
|  19 | INPUT      |      | 0 | 35 || 36 | 0 |      | INPUT     |  16 |
|  26 | INPUT      |      | 0 | 37 || 38 | 0 |      | INPUT     |  20 |
|        GND                  | 39 || 40 | 0 |      | INPUT     |  21 |
+-----+------------+------+---+----++----+---+------+-----------+-----+
| BCM |    Name    | Mode | V |  Board   | V | Mode | Name      | BCM |
+-----+------------+------+---+  Pi 4B   +---+------+-----------+-----+

rondoc, 2024-11-14:
> (venv) ***@***.***:~/kiln-controller $ ./test-thermocouple.py

Make sure the thermocouple leads are NOT grounded.

BlakeCLewis, 2024-11-14:
32f is 0 volts, so grounded or maybe an open/loose connection(broken wire or thermocouple). I have had issues with heat cycles (expansion and contraction) loosening up connections. The connections looked good, but a turn of the screws to tighten them up helped. If you have a way to measure mVolts, 70f should be about 1 to 1.5 mV. I do not use this project. I have referenced it while working on my project.

wasexton, 2024-11-15:
Thank you both for the responses.  I have validated that neither lead is grounded and that the connections are tight.  I am going to try to borrow a meter that can check mV as my current analog one (ancient) is not that sensitive.  Any other suggestions as I think I have everything connected properly? When I didn't I was getting errors and now I am just getting a static temperature.

rondoc, 2024-11-15:
Connections using the pins on the board are not always reliable. Try
rewiring with new .

wasexton, 2024-11-15:
Good idea!  I was also routing through a circuit board in order to add the resistors.  I will try direct wiring.  Though I did a continuity test from the bottom of the pin on the Pi to the back of the pin on the 31855 and it was good.

wasexton, 2024-11-15:
That worked!  Thank you so much! I want to rewire and see if I made a bad connection with the resistors or if something else such as a loose connection on one of the pins was the culprit.

rondoc, 2024-11-15:
Glad it worked!

closed 2024-11-16 by jbruce12000