jbruce12000/kiln-controller #202: Support for Pi 5?

brentintipp, 2024-11-22:
I've been trying to get this running on a Pi 5.  I know at the time this was created, Blinka didn't support the Pi 5.  However, I believe Blinka has been updated to support the Pi 5.  The problem is that the GPIO control has changed on the Pi 5.  Supposedly, you can use gpiod or gpiozero to interface with the pins on the Pi 5.

Gpiozero appears to be much more user friendly.  I am able to control one of the pins from a Python script to turn the kiln on and off.  However the update of all of the code to get the kiln controller working is beyond my skills.

If I understand correctly, gpiozero will work with the older boards in addition to the Pi 5 and other SBC's.   Would it be reasonable to update this to use gpiozero?  I thought of forking a project for this purpose in the hope of getting others to help update the code.  Is that reasonable?  Any advice?

jbruce12000, 2024-11-30:
so you tried https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi

and could not get your pi5 to work?

This library is written by adafruit. They release a new version of blinka when they release new boards. 

There is no reason not to use gpio zero... read here...
https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/circuitpython-raspi
even the adafruit folks advocate for it. I chose adafruit so I would not longer have to manage either single board computer gpio connectivity or support of tc hardware like the 31855 or 31856. It's been a mixed bag for me because the coders at adafruit are not consistent in the way they handle errors, so I had to make up for that on my end.

The PI5 was added to detection in adafruit blinka on Oct 6th 2023 here...
https://github.com/adafruit/Adafruit_Python_PlatformDetect/commits/main/adafruit_platformdetect/revcodes.py
and every PI ever made is supported.

I think I need to better understand the problem you're running into.

brentintipp, 2024-12-02:
> so you tried https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi
>
> and could not get your pi5 to work?
>
> This library is written by adafruit. They [...]

You are correct, Blinka now supports the PI5.  I'm not an expert on this stuff, but as I understand it, the problem is that the PI5 requires a different library than older devices due to the new chip controlling the GPIO pins.  Specifically, RPi.GPIO will not work, it requires libgpiod or gpio zero.  I'm not sure if it would be as simple as changing the line in Requirements.txt from RPi.GPIO to libgpiod.  I was merely suggesting that if the kiln-controller code was modified to use GPIO zero, then it might work with the PI5 and also be backward compatible with the older Pi SBCs.

In any case, I installed your kiln-controller code on a PI4b and have it working.  Thanks!

jbruce12000, 2024-12-03:
interesting. I don't own any PI5s, so I cannot test it. Does anyone else have a PI5 and interest in trying it out?

jbruce12000, 2024-12-08:
see [this thread](https://github.com/jbruce12000/kiln-controller/issues/201#issuecomment-2526275706) for getting a version of RPi.GPIO that works with the PI5.

closed 2024-12-08 by jbruce12000