jbruce12000/kiln-controller #204: Thermocouple not connected

fIuffyparrot, 2024-12-18:
Raspberry Pi4 model B
Max31856
K type thermocouple

Pins:
Sck -> GPIO11 SPI_CLK (pin 23)
Sdk -> GPIO9 SPI_MISO (pin 21)
Sdi -> GPIO10 SPI_MOSI (pin 19)
Cs -> GPIO5 (pin 29)

This is my first time using a raspberry pi. I have some experience wiring small electronics and I have written python scripts (albeit the scripts I write are only used for GIS software). I am hoping I've done something that is obviously wrong to someone here as I am out of ideas. I have spent the last 3 days trying to get the ./test-thermocouple to work. I only get a message saying "thermocouple not connected". I have ensured that everything is wired properly, has good connections etc. I updated the config file to reflect my hardware, but when i run the test thermocouple command in a terminal, text comes up showing that gpios and thermcouple are still reading as the default options despite me changing them. Is there another config file I'm missing or something?

It looks to me like the test-thermocouple.py isn't actually reading the config.py. I checked for error logs and there are none. Let me know what information I need to provide to figure this out and I will reply as soon as I am able to. Any help is appreciated.

fIuffyparrot, 2024-12-18:
Adding more info that I think is relevant, I was tired when I wrote the original post.

Thermocouple has one pole connected to the positive terminal and the other pole connected to the negative terminal on the max31856. I'm mentioning this as I've seen people saying to not ground the the tc. Every photo I've seen shows the tc grounded the way I have mine and I don't see how it would function otherwise. I'm guessing what everyone means is not having the tc touch anything else that is conductive. 

I have rewired to different gpio pins a couple of times. I was a bit confused when I found that the max31856 pin labels were different from the max31855. I had to look around to see how others wired the 56 and landed on the pin selection mentioned above. I am fairly sure I have it correct, but maybe this is where I am going wrong. Results were the same on different setups anyway.

As I said previously, I went over how everything was wired a few times. I used a multimeter to make sure everything has a good connection. Nothing is shorting and I get voltage where I would expect it to be. 3.3v to the 31856. 

I am leaning towards there being some kind of issue with the py files. I previously mentioned that the test-thermocouple script wasn't reflecting the changes I made to the config file. Here are the changes I made to reflect my hardware:

try:
    import board
    spi_sclk  = board.D11    #spi clock
    spi_miso  = board.D9    #spi Microcomputer In Serial Out
    spi_cs    = board.D5    #spi Chip Select
    spi_mosi  = board.D10    #spi Microcomputer Out Serial In (not connected) 
    gpio_heat = board.D23    #output that controls relay
    gpio_heat_invert = False #invert the output state
except (NotImplementedError,AttributeError):
    print("not running on blinka recognized board, probably a simulation")

And

max31855 = 0
max31856 = 1
# uncomment these two lines if using MAX-31856      (<- This comment confuses me as the two lines below it were already uncommented)
import adafruit_max31856
thermocouple_type = adafruit_max31856.ThermocoupleType.K

As I understand it, this should have the config file set to use the max31856, but when I run the test-thermocouple command I get a line that states "thermocouple: adafruit max31855". This would be from lines 51-59 in the test-thermcouple.py. It looks like despite having the config file set up as I do, the test-thermocouple.py is still looking for a max31855. So I suspect I have to be doing something wrong in the config.py. Problem is I don't see anything else in the config that references which board I am using. It looks like older versions imported a max31856 library instead of blinka. I tried writing in the import line that was used in the past to see if that would change anything (it didn't). I commented out that import line I added when it didn't change anything. I hope the extra information helps someone figure out what I am doing wrong. Thanks everyone.

rondoc, 2024-12-18:
1. Grounding..the thermocouple leads are fully floating NOT grounded. That
means anywhere ! Check with meter and make neither lead is grounded.  2.
Suggest using the max31855 ada fruit board and set up in the config.py to
match. I have been using 55 board and K thermocouples for several years
with complete success.
The reason for thermocouple problems is nearly always a simple wiring error
not a major software bug.

fIuffyparrot, 2024-12-20:
When you say to not ground the thermocouple, do you mean I’m not supposed to connect one of the leads to the negative terminal of the max31856 board? I’ve tried running it with the negative terminal connected and disconnected and the results were the same. The tc is not grounded to anything else. 

I don’t mean to imply I’m blaming the software. I believe I am doing something wrong with the software. When I run the test-thermocouple command, the board and pins in use are printed before it starts reading the tc output. I set my pins in the config file but the print just reads whatever the original settings were anyway. I think the script is looking for board/pin settings that I don’t have. 

I ordered a max31855 yesterday. Would’ve done that to begin with if I knew the 56 would be more trouble. Without any other knowledge I figured I should go with the 56 so I would have more options down the road.

fIuffyparrot, 2025-01-15:
I am updating this in case anyone else is running into similar issues. Hopefully this info helps someone.

1. If you are like me and just want a functioning kiln, don't bother with the Max31856. When I started, I saw that the 31856 allowed the use of multiple thermocouple types. Without further information that seemed like the obvious better choice. If you need to use something other than a K type thermocouple then get that board. But if just need this setup to work, get a Max31855. The 31856 seems to be a bit more complicated and the config file wasn't playing nice. Odds are I didn't have the config file set up correctly. The 31855 is nearly plug and play.

2. After I got the Max31855, I kept getting a 32* F reading. If you get this issue, odds are you will be told to make sure the thermocouple is not grounded. In my case, the issue was the Max31855 board itself. I initially bought mine on Amazon. I was a bit hastey and didn't notice that the one I grabbed was a knock off. Buy your board directly from Adafruits website. Once I got the legit Max31855, I swapped it out and it immediately started working without changing any other factors.

closed 2025-01-15 by fIuffyparrot

rondoc, 2025-01-15:
Beware of knock off boards because they might be using a defective chip.
For the difference in $$ and the waste of your time it is just not worth it.
Ronald ( 80 years of electronics. !!)

TinMaan, 2025-01-18:
I am glad you solved the problem. I've had the same issue with the 31856
board. Then I switched to a knock-off 31855 from Amazon with another set of
failures. To this point, I abandoned my project in frustration. As you
recommended I've purchased one of 31855 from Adafruits to pick up the
project again.
Are you able/willing to share a copy of your working config.py? Otherwise
perhaps you could share your entries on some specific area. I am a
completely nooob in the world of RaspberryPi

fIuffyparrot, 2025-01-18:
I don't know how to reply directly to a comment so hopefully you see this.

This is also the first time I've ever touched a raspberry pi. I have a tendency to get into projects that I know nothing about. But I'm stubborn and usually I figure it out eventually. I'm at the point where I need to actually connect the pi to my kiln and start tuning it. I am waiting on my friend whos an electrician so I don't kill myself or break something.

Here is how my config file is set up.

    spi_sclk  = board.D22    #spi clock
    spi_miso  = board.D17    #spi Microcomputer In Serial Out
    spi_cs    = board.D27    #spi Chip Select
    spi_mosi  = board.D10    #spi Microcomputer Out Serial In (not connected) 
    gpio_heat = board.D23    #output that controls relay
    gpio_heat_invert = False #invert the output state

This isn't how it is set up by default but it is working for me. Connect the pins to the Max31855 exactly as it is shown in the schematics image on the main page. Here is another guide I found useful for some parts despite this guy using a Max31856: https://www.hylkerozema.nl/2024/04/27/raspberry-pi-controlled-ceramic-kiln/
(There are two parts to that guied)

If you name your Pi something else like I did, you'll need to change that name in a couple of files (it's "pi" by default). I don't remember where these files are off the top of my head but someone mentions this in the issues section of this project. 

I recommend buying your thermocouple directly from the link provided on the main page. I bought my first one from Amazon and it arrived cracked due to poor packaging. It also did not have an easy way to connect wires to it. The one on the main page arrived well packaged and has easy connection spots for wiring as well as a way to fix the tc to your kiln securely. 

It is probably ideal to buy your SSR from the link, but I got away with an Amazon SSR. Just make sure you get one that uses a DC input. I wasn't paying attention and originally bought one that used an AC input and the Pi wasn't putting out enough current to fully activate it. 

I was thinking about writing a build guide for a layman. I think it would be useful for someone that has never used a Pi before. Let me know if you have questions, I'll help as best I can.

TinMaan, 2025-01-19:
Thank you for this detailed information. The renaming Pi definitely
would've been another one of those " WTF " moments for me in this whole
setup.
One major confusion I have in the config.py settings is in the "
thermocouple breakout boards " section ( line 97 - line 106 ). The default
settings appear to be for a 31856 board as lines 105 and line 106 are
un-commented. Meanwhile there is no " import " command for 31855. So, if we
are using a Max31855 board, should the lines 105 and 106 be commented?
Yes, I wish somebody would create a layman build guide for this system. As
a matter of fact I would pay a fair amount for that work. The included
README.md is helpful but it's far from complete.

Thanks again for taking time to reply to my cry for help.

fIuffyparrot, 2025-01-19:
I did not change anything from lines 102-106. The py files can import adafruit_max31856 if it wants to, but it's going to look for a max31855 since that is set to '1' as seen below. When I was trying to make the 56 board work these lines were confusing. It says to "uncomment these two lines if using MAX31856". But they are already uncommented. And even when I set 55 to 0 and 56 to 1, the test-thermocouple.py would not react to this change and continued to look for a 55. This is one of the reasons I gave up on using a 56 board and went with the easier option. 

max31855 = 1
max31856 = 0
# uncomment these two lines if using MAX-31856
import adafruit_max31856
thermocouple_type = adafruit_max31856.ThermocoupleType.K

I will put together a written guide once I have a working and tested kiln going.

TinMaan, 2025-01-20:
Great, thanks for the update. I will start testing as soon as I receive my
Max31855.
I like your idea for getting a high voltage electrician involved.
Specifically I prefer to keep the Kiln-Sitter and also the good ol' timer
unit as disaster prevention and that is definitely a job for an
electrician. I can't trust the kilns running on one kind of controller
without failsafe systems onboard. Perhaps a simple Arduino or Pi program
could run a separate mechanism to cut the mains power in case of a runaway
kiln.
I am looking forward to your guid.

BlakeCLewis, 2025-01-21:
I do not use this project, but I use Python, max31855 and max31856, ktype,
on a pi.
There are 2 spi hardware buss, 0 and 1.
It looks like you are using SPI0, implemented in hardware. I think your
chip select (CS) is wrong.
---- SPI0 pins ---
SCLK 23
MISO 21
MOSI 19
CS0   24

'''------------------- begin python script
------------------------------------------------------------
 Here is my test script. It used an LCD screen.
 I remove the LCD stuff and replace with print() which will show up on the
console.
 I have not used this in years, but I am pretty sure it will work.
 You may not need 'from signal import *'.
'''
from signal import *
import os
import time
from Adafruit_GPIO import SPI
from Adafruit_MAX31856 import MAX31856

'''
  When using the hardware spi you do not set which pins are connected.
  You basically just tell the driver which bus and chip select'
SPI_PORT = 0
SPI_DEVICE = 0
sensor = MAX31856(hardware_spi=SPI.SpiDev(SPI_PORT, SPI_DEVICE))

while True:
  temp = sensor.read_temp_c()
  internal = sensor.read_internal_temp_c()
  print('Thermocouple Temperature: {0:0.3F}*C'.format(temp))
  print('    Internal Temperature: {0:0.3F}*C'.format(internal))
  print()
  time.sleep(5)

'''----------------------- end of script
---------------------------------------------------------'''

k-type pro tip: one side is magnetic. The k-type wire is made of same
material as the thermocouple. Test with magnet. Hook magnetic wire up to
magnetic side of thermocouple.

If using software 'bitbanging', most any GPIO pin will do. Software is
slower and hardware is faster/easier

The max31856 does cold junction compensation, polynomial correction, and
filters out 50/60hz AC line noise. The max31855 does not do the polynomial
correction.

  sudo raspi-config
  #enable interfaces spi

  git clone https://github.com/adafruit/Adafruit_Python_GPIO.git
  git clone https://github.com:johnrbnsn/Adafruit_Python_MAX31856.git

It has been 6 years since I touch the code. Still working well.

BlakeCLewis, 2025-01-21:
missed some quotes right before the SPI_PORT, here is fixed
'''------------------- begin python script
------------------------------------------------------------
 Here is my test script. It used an LCD screen.
 I remove the LCD stuff and replace with print() which will show up on the
console.
 I have not used this in years, but I am pretty sure it will work.
 You may not need 'from signal import *'.
'''
from signal import *
import os
import time
from Adafruit_GPIO import SPI
from Adafruit_MAX31856 import MAX31856

'''
  When using the hardware spi you do not set which pins are connected.
  You basically just tell the driver which bus and chip select
'''
SPI_PORT = 0
SPI_DEVICE = 0
sensor = MAX31856(hardware_spi=SPI.SpiDev(SPI_PORT, SPI_DEVICE))

while True:
  temp = sensor.read_temp_c()
  internal = sensor.read_internal_temp_c()
  print('Thermocouple Temperature: {0:0.3F}*C'.format(temp))
  print('    Internal Temperature: {0:0.3F}*C'.format(internal))
  print()
  time.sleep(5)

On Tue, Jan 21, 2025 at 1:59 PM Blake Lewis ***@***.***> wrote:

> I do not use this project, but I use Python, max31855 and max31856, ktype,
> on a pi.
> There are 2 spi hardware buss, 0 and 1.
> It looks like you are using SPI0, implemented in hardware. I think your [...]

-- 
Blake C. Lewis

TinMaan, 2025-02-15:
Houston we have a liftoff!
As a matter of fact even one of cheapest knockoffs of MAX31855 from
AliExpress ( $3 ) works with your settings and the main wiring diagram
shared on Github. To save time, I purchased a few copies of MAX31855 from
various sources and basically they all worked.
Now to the high voltage / high current setup. I sent the original wiring
diagram of my kiln to the electrician with a detailed reminder of the work
at hand. His first reaction : The discounted version of work would entail,
putting all the resistive elements-sets in parallel and controlling them in
unison with * two * SSR on each hot leg of the 240V mains. The not very
discounted version would add up to the cost of a Bartlett Genesis v2. So, I
took the door #1 naturally.
Now I really like to know why the originator of this project ( J. Bruce !?
) never mentioned the need for controlling both hot legs of 240V powering
our kilns. Perhaps he fired his kiln with 120V.
I will update as soon as the electrician actually powers the kiln.

Thank you again for taking the time and sharing your knowledge. I couldn't
complete this project without your help.

rondoc, 2025-02-16:
> Houston we have a liftoff! <br> As a matter of fact even one of cheapest
> knockoffs of MAX31855 from <br> AliExpress ( $3 ) works with your settings
> and the main wiring diagram <br> shared on Github. To save time, I
> purchased a few copies of MAX31855 from <br> various sources and basically
> they all worked. <br> Now to the high voltage / high current setup. I sent
> the original wiring <br> diagram of my kiln to the electrician with a
> detailed reminder of the work <br> at hand. His first reaction : The
> discounted version of work would entail, <br> putting all the resistive
> elements-sets in parallel and controlling them in <br> unison with * two *
> SSR on each hot leg of the 240V mains. The not very <br> discounted version
> would add up to the cost of a Bartlett Genesis v2. So, I <br> took the door
> #1 naturally. <br> Now I really like to know why the originator of this
> project ( J. Bruce !? <br> ) never mentioned the need for controlling both
> hot legs of 240V powering <br> our kilns. Perhaps he fired his kiln with
> 120V. <br> I will update as soon as the electrician actually powers the
> kiln. <br> <br> Thank you again for taking the time and sharing your
> knowledge. I couldn&#39;t <br> complete this project without your help.
> <br> <br> On Sun, Jan 19, 2025 at 2:12 PM fIuffyparrot ***@***.***&gt; <br>
> wrote: <br> <br> &gt; I did not change anything from lines 102-106. The py
> files can import <br> &gt; adafruit_max31856 if it wants to, but it&#39;s
> going to look for a max31855 <br> &gt; since that is set to &#39;1&#39; as
> seen below. When I was trying to make the 56 <br> &gt; board work these
> lines were confusing. It says to &quot;uncomment these two <br> &gt; lines
> [... cut ...]
> option. <br> &gt; <br> &gt; max31855 = 1 <br> &gt; max31856 = 0 <br> &gt;
> uncomment these two lines if using MAX-31856 <br> &gt; <br> &gt; import
> adafruit_max31856 <br> &gt; thermocouple_type =
> adafruit_max31856.ThermocoupleType.K <br> &gt; <br> &gt; I will put
> together a written guide once I have a working and tested kiln <br> &gt;

There are not two.legs in a single phase 240 volt system . Therefore
younonly need one Triacs orvtwo scrs back to back. Or you can buy a
module.  They will all work in this system.
Ronald