jbruce12000/kiln-controller #205: Question about SSR's vs contactors

Nikodemuzz, 2025-02-16:
Hello!

I have an old ceramics kiln with a very rudimentary control system, and I would benefit greatly by using this solution for controlling the kiln instead!

However, the old kiln (400V) uses traditional contactors for switching power on and off. If possible, I would like to keep that solution for three reasons: 

1. The contactors work fine 
2. The contactors work better in high heat environments. My kiln heats up to 1300C. Going with SSR would require new cooled housings or new wiring for a more remote installation.
3. Cost savings

My question is (as a non-coding person), can the controller work also with contactors?

Thank you!

Nikodemuzz, 2025-02-16:
Thanks a lot for your quick reply, Ronald!

Yeah, that's true I need something in between to handle contactors, like power mosfets. I was more concerned about the switching in the code. If it is as simple as making the switching time longer I'm sure I can handle that as long as I can find the place where to change it! =)

The contactor I have is a Lovato MC9.F, an Italian design. I didn't find anything on this particular model on the internet, but I'm pretty sure it is a typical 3 phase contactor, rated for 16A and 600V.

It might be a good idea to change to SSR like you say, and I might still do that once I have the facts for both cases laid out.

I think your email was censored by GitHub, I see just ***@***.***

BlakeCLewis, 2025-02-16:
I use relays with DC 12v coil. I run the 5v from the pi to a uln2803a (about $1 USD) and it switches the 12v. I think it can do 7 channels. I use 4 channels, 1 for 12v electronics cooling fan, and 1 for each relay. There is a relay for each of the 3 sections of the kiln. The controller has a 30 second cycle. The kiln has done about 100 firings (120000 cycles). The big question: what does the coil need? From the spec sheet, I see it is offered with AC or DC with different voltages.

rondoc, 2025-02-16:
One big advantage with the SSR is you can get zero crossing SSR. This means
almost no RFI from sparking contacts.
You said it was a 3 phase contactor. Are you using a 3 phase supply for the
kiln?

chipgarner, 2025-02-16:
@Nikodemuzz  I can help with the software if needed.

Nikodemuzz, 2025-02-16:
> One big advantage with the SSR is you can get zero crossing SSR. This means
> almost no RFI from sparking contacts.
> You said it was a 3 phase contactor. Are you using a 3 phase supply for the
> kiln?
> […](#)
> 
>  I use relays with DC 12v coil. I run the 5v from the pi to a uln2803a
>  (about $1 USD) and it switches the 12v. I think it can do 7 channels. I use
>  4 channels, 1 for 12v electronics cooling fan, and 1 for each relay. There
>  is a relay for each of the 3 sections of the kiln. The controller has a 30
>  second cycle. The kiln has done about 100 firings (120000 cycles). The big
>  question: what does the coil need? From the spec sheet, I see it is offered
>  with AC or DC with different voltages.
> 
>  <[#205 (comment)](https://github.com/jbruce12000/kiln-controller/issues/205#issuecomment-2661555887)>,
>  ([jbruce12000/kiln-controller#205](https://github.com/jbruce12000/kiln-controller/issues/205))
>  <[#205 (comment)](https://github.com/jbruce12000/kiln-controller/issues/205#issuecomment-2661555887)>
> 
>  I use relays with DC 12v coil. I run the 5v from the pi to a uln2803a
>  (about $1 USD) and it switches the 12v. I think it can do 7 channels. I use
>  4 channels, 1 for 12v electronics cooling fan, and 1 for each relay. There
>  is a relay for each of the 3 sections of the kiln. The controller has a 30
>  second cycle. The kiln has done about 100 firings (120000 cycles). The big
>  question: what does the coil need? From the spec sheet, I see it is offered
>  with AC or DC with different voltages.
> 
>  <[#205 (comment)](https://github.com/jbruce12000/kiln-controller/issues/205#issuecomment-2661555887)>,

Yes, the kiln uses 3 phase power. 16A, 400V.

Nikodemuzz, 2025-02-16:
> I use relays with DC 12v coil. I run the 5v from the pi to a uln2803a (about $1 USD) and it switches the 12v. I think it can do 7 channels. I use 4 channels, 1 for 12v electronics cooling fan, and 1 [...]

This is a good question, on the part it doesn't say and I can't find any documentation. If I had to guess I would say 12V. It would of course be possible to measure what comes out of the current control box

Nikodemuzz, 2025-02-17:
I will measure the contactor coil voltage next time I'm at the workshop, probably later this week.

Meanwhile, I got to thinking about the time cycle of the controller. Surely that is something that needs to be adjusted based on desired performance? For example, the current controller will have the heating on say, 5 seconds and wait for 1 minute in the beginning of the program where the heating rate is really slow. Towards the end of the program it will pretty much keep the heating on constantly to reach the target temperature. I got slightly worried that if there is a set pulse time that the controller uses and then just varies the wait time in between, I might not be able to make it work.

I tried to check the corresponding part in the code to try to understand how it works but didn't find it.

rondoc, 2025-02-17:
You are correct. If you cannot have short gaps between on and off then the
code needs to change as at max temps you probably need heat on 100%
continuously. Correct?
My original code handled the cycling in a different way than the pi one. It
kept a duty cycle that changed with the temperature so as the temperature
rose in the kiln the duty cycle increased. This works well but I changed to
the pi system mostly because I like the android app that allows me to
monitor remotely.
A combo of the pi code and mine would IMNSHO be better than the fixed duty
cycle we have now. May be I can get AI to write the code!!
Ron.

Nikodemuzz, 2025-02-17:
Yes, exactly! Reaching max temps would not require 100% heating necessarily, but close to it. The option should be there also for programs that calls for fast heating rates. Changing the duty cycle as a function of temperature is exactly the solution for that problem, since with a fixed cycle time I don't think it will be possible to balance the need to achieve slow, controlled heating rates and not causing unnecessary cycling of the contactor. In a larger kiln this might be compensated with larger thermal mass, but mine is quite small.

So unless AI rides to the rescue I guess I will have to look at the SSR route after all. =) Would you recommend installing the relay away from the kiln in order to have better cooling? The contactor is mounted on the kiln itself, which does get quite hot on the surface while running.

BlakeCLewis, 2025-02-17:
Pictures I found with that relay show a small label on the back of the
relay with coil voltage. I think the coil on the relay is a module and the
module has label on it. Or, as you said, 'measure it'.
If it is 12v the data sheet said 4/4w. 4w start / 4w hold. so 1/3 amp. I
think the chip I used can do 1/2amp

rondoc, 2025-02-17:
> <br> relay with coil voltage. I think the coil on the relay is a module and
> the <br> module has label on it. Or, as you said, &#39;measure it&#39;.
> <br> If it is 12v the data sheet said 4/4w. 4w start / 4w hold. so 1/3 amp.
> I <br> think the chip I used can do 1/2amp <br> <br> <br> On Sun, Feb 16,
> 2025 at 3:32 PM Nikodemuzz ***@***.***&gt; wrote: <br> <br> &gt; I use
> relays with DC 12v coil. I run the 5v from the pi to a uln2803a <br> &gt;
> <br> &gt; 4 channels, 1 for 12v electronics cooling fan, and 1 for each
> relay. There <br> &gt; is a relay for each of the 3 sections of the kiln.
> The controller has a 30 <br> &gt; second cycle. The kiln has done about 100
> firings (120000 cycles). The big <br> &gt; question: what does the coil
> need? From the spec sheet, I see it is offered <br> &gt; with AC or DC with
> different voltages. <br> &gt; <br> &gt; This is a good question, on the
> part it doesn&#39;t say and I can&#39;t find any <br> &gt; documentation.
> If I had to guess I would say 12V. It would of course be <br> &gt; possible
> to measure what comes out of the current control box <br> &gt; <br> &gt; —
> ID: <br> &gt; ***@***.***&gt; <br> &gt; [image: Nikodemuzz]*Nikodemuzz*
> left a comment <br> &gt; (jbruce12000/kiln-controller#205) <br> &gt; &lt;
> <br> &gt; <br> &gt; I use relays with DC 12v coil. I run the 5v from the pi
> to a uln2803a <br> &gt; (about $1 USD) and it switches the 12v. I think it
> can do 7 channels. I use <br> &gt; 4 channels, 1 for 12v electronics
> cooling fan, and 1 for each relay. There <br> &gt; is a relay for each of
> the 3 sections of the kiln. The controller has a 30 <br> &gt; second cycle.
> The kiln has done about 100 firings (120000 cycles). The big <br> &gt;
> <br> &gt; with AC or DC with different voltages. <br> &gt; <br> &gt; This
> <br> &gt; documentation. If I had to guess I would say 12V. It would of
> course be <br> &gt; possible to measure what comes out of the current
> ID: <br> &gt; ***@***.***&gt; <br> &gt; <br> <br> <br> -- <br> Blake C.
> Lewis <br>

Even with a SSR you might find at the top end the time might be longer
because although the pi will say 100% it's not on full. Will someone
correct me if I'm wrong.you can't have 100% and a off time..
Ron.

jbruce12000, 2025-02-23:
Hi @rondoc. The code, as written today, if the PID says heat 100%, does not switch from heat to cool every duty cycle.

It's a rare event for the PID to push the output to 100% if the PID is within the PID control window. This usually happens only if the schedule is very aggressive and the kiln elements cannot keep up, or if the elements are very old.

Outside the pid control window, going to 100% happens a lot depending on how _throttle_below_temp_ and _throttle_percent_ are set.

rondoc, 2025-02-23:
Hi can you elaborate how throttle works?
If i wanted to change the settings where is the piece of code. I have
noticed that after I reach max temp set and want the temp to decline very
slowy ( still needs power input) . (  As i have set it graphically) The PID
doesn't follow as it does elsewhere but the decline is faster.
Thanks
Ronald.

Anger is usually regrettable, kindness rarely.

jbruce12000, 2025-02-28:
These settings are in config.py at the very end...

########################################################################
# low temperature throttling of elements
# kiln elements have lots of power and tend to drastically overshoot
# at low temperatures. When under the set point and outside the PID
# control window and below throttle_below_temp, only throttle_percent
# of the elements are used max.
# To prevent throttling, set throttle_percent to 100.
throttle_below_temp = 300
throttle_percent = 20

So, if you're kiln elements don't cause overshoot at low temps, just set throttle_percent to 100.

rondoc, 2025-03-01:
Thanks.

Anger is usually regrettable, kindness rarely.

digiexchris, 2025-03-25:
Can always use the SSR to trigger the contactor. You can get them in AC output, DC output, high voltage AC, high voltage DC, etc..

I use a contactor as a backup in case my SSR fails in the on state. I have a separate circuit watching for a runaway, and it cuts input power via contactor. A separate SSR does that. Piddly little 240v 15A contactor though so the SSR to trigger the coil was inexpensive. The coil voltage was line voltage though so the SSR made sense (instead of something cheaper like a power mosfet)

rondoc, 2025-03-25:
I have the same SSR in use since 1987 and it works fine.
Ronald.

Anger is usually regrettable, kindness rarely.

digiexchris, 2025-03-25:
> I have the same SSR in use since 1987 and it works fine.
> Ronald.
> 
> Anger is usually regrettable, kindness rarely.
> […](#)

apologies for the unclear message, I was mentioning an example of using an SSR to trigger a contactor, I wasn't making a recommendation for a specific configuration. Given that low current SSRs with a high enough voltage range to trigger an AC contactor are common and low cost, it might be a reasonable solution for the original topic.

Also, if you are commenting on reliability, past performance does not indicate future performance on critical (or heck, most) equipment, particularly when we consider age generally degrades reliability, not improves it. just saying ;) Though a runaway in my case would be safe, it would just burn out the coils up past 1500C instead of escaping the enclosure, so I guess it's self limiting in two ways. I'd prefer to not lose the coils if the primary SSR latches on though, which happens to be the most common failure mode for TRIACs and SCRs (short circuit failure mode is way more common than failing open).

rondoc, 2025-03-26:
Scr 's fail mostly through voltage spikes. These punch through the layers
and a short results. If you want a reliable SSR I recommend using modules
designed for industrial applications. For a 240 volt system use 600 volt as
the spec. Also heat sinking us critical . My very long time experience with
electronics does mean that my advice has some value.
Mechanical relays don't work for long when switching frequencies are high.
The contact materials and spacing are also critical. The relative cost for
contractors and SSR s is wide because
of manufacturing processes. By all means use a contactor for over temp
cutoff. I prefer to monitor my kilns when nearing max temps so I can always
manually use a breaker. Never have had to so.

digiexchris, 2025-03-26:
yup, indeed, good points to consider. I too have never had an SSR fail, but I've had many many relays fail. I figure, as long as people are thinking about the "what if"'s, they're FAR safer than anyone that hasn't considered it.

Sorry for going even more off topic, but do you have any experience with a zero-crossing SSR operating with a frequency near or greater than the line frequency? I have one that's doing 1khz PWM in a different application, but I haven't bothered putting an oscilloscope on it yet. I suspect it's actually capping itself at line frequency due to the zero crossing functionality. I wonder if that generates excess heat or problems, but it's been running for 7 years that way.

rondoc, 2025-03-26:
The zero crossing modules are designed for line frequency application. I
have not used one for a different frequency. It would have to a sine or
trap. wave to allow the timing to be ok.
My kiln controller built with discrete chips used a zero crossing chip and
by choice could pick as many cycles as I want from 0 to 16 . So I have 16
levels of input all synchronized to the line.
If you using a SSR to switch high frequency square waves that's a clear
mis-use and the result is not predictable. Might be interesting to try
though!
Ronald.

Anger is usually regrettable, kindness rarely.

mar2hall, 2025-05-08:
I am also setting up the kiln-controller converting an old large Naber kiln that uses 3-Phase, using an SSR to activate a 3-phase contactor like @digiexchris mentioned.

Was there a definitive answer on which config settings to alter to reduce the amount of switching of the mechanical contactor? 
Still getting my head around how the system works and the terminology but I believe sensor_time_wait just slows down the reading of the thermocouple, but does not slow down the pid cycle time.

Is there a variable that slows down the pid cycle time?

digiexchris, 2025-05-22:
@mar2hall sensor_time_wait is what you want. it is the window in which the system decides to take action, ie. pid cycle window.

Nikodemuzz, 2025-06-15:
I want to thank everyone who participated in the discussion, lots of helpful and useful information was shared! Even though I think I would now understand how to make the controller work with contactors, I ended up purchasing a solid state relay and will be moving forward with that!

Nikodemuzz, 2025-06-16:
That is a good point about the safety of the SSR. I was considering that, since I have the contactor at hand anyway. A runaway kiln that cannot turn itself off is pretty much the worst scenario that one can have with this kind of a device, and it should be accounted for somehow especially when the probability of it occurring isn't vanishingly low. 

So basically it could be so that on startup of the kiln profile the contactor is engaged, and disengaged when the program ends, or upon failure. I guess the software does not accommodate this right now, but it would have to be coded in?

Related to the above is temperature management of the SSR. There is a fan on the heatsink of my SSR, and I would like to implement also some case fans to my installation to ensure proper airflow. Has anyone implemented a fan controller within this kiln controller, or would it make sense to look for a standalone solution? I don't think this should have to be anything fancy, just turn on the fans when the system is powered up, basically.

rondoc, 2025-06-17:
> @Nikodemuzz <https://github.com/Nikodemuzz> I have 12v coming into my
> electronics box on the side of my kiln for various reasons, one of which is
> running common computer case fans. Then it goes into a buck/boost converter
> to drop it to 5.1v for the rpi to run off of. I have it wired so that when
> the power switch is on, the fans are running regardless of the state of the
> SSRs, since they're quiet fans anyway.

I see no reason to use anything but K type TCS for ceramics. They are cheap
strong and reliable upto cone11. Then use the 55 chip as it is also cheap
and reliable. Also make sure that the tc wires are insulated and don't
connect to ground.  The KISS method works ..
Ronald.

Anger is usually regrettable, kindness rarely.

closed 2026-10-10 by jbruce12000