jbruce12000/kiln-controller #207: Setting a temp without a profile?

digiexchris, 2025-03-23:
Is there a way to turn the kiln on and have it heat to a set point? I use it for normal kiln type ramp and soak stuff, but also need a manual mode where I control the temp myself for metal casting purposes. Is that possible?

rondoc, 2025-03-30:
I would make a profile that had the target temp for as long as wanted.
Don't see why that wouldn't work .
You could easily make several "one temp" profiles and then goto them at
will.
Ronald.

digiexchris, 2025-03-31:
I thought about this a little more, and I think I can explain my needs better. I very likely am just missing something with how it works.

it works great on my burnout furnace where the process is decided on from the start and I can let it run, but melting his highly variable and I need to adjust to respond to various factors while I watch.  I was hoping for a feature to allow a quick access to editing the current set point or temporary modification to the current set point. Kind of like how you can adjust the temperature of a 3d printer while it's printing without stopping and re-slicing the job. Maybe it exists and I'm just blind?

The profiles are still handy for melting because I need to ramp up to temp slowly to avoid blowing up my insulation, and also for heat treating metals later on, so a kiln type controller is still valuable. But once at melt temp I need to bump the temp up or down as things progress, not according to a schedule or a particular alloy.  I'd literally need a profile every 25 degrees from 600C to 1350C (for alloys throughout the range, and as the melting point changes as alloys are created), to avoid too much increased workload with the phone while also managing the melt and other associated tasks. 

Granted, this is not a typical use case for a kiln, and so I understand this software isn't designed for it, but it would be a nice addition since there isn't really any good alternatives for metalworking furnaces. How hard would it be to add a different frontend that can be switched to that's more specific to semi-manual operation like this?

rondoc, 2025-03-31:
My controller will do what you want manually. It can provide 16 steps in
power from 0 so you can just I increment or decrement by whatever power you
want. A bit like having a digital rheostat. I could add an app so you could
monitor and control it remotely but this getting away from this site's
purpose.  So if interested email me directly at ***@***.***

digiexchris, 2025-04-04:
Well, I could do the same thing with a simple PWM controller in that case and not use a full blown controller at all. I do need the PID temp control for maintaining precise temps. A digital kiln controller has all of the underlying functionality required, I may just have to write an addition to the UI.

rondoc, 2025-04-04:
A simple pwm is not good because you will be switching during the full
cycle. Too much rfi!
My system uses the full cycle but you select how many cycles . This gives
16 levels of power. I could do that with a simple Arduino  and chatgpt can
write the sketch in seconds!
I did it with discrete logic chips but that was in 1987!
Ronald

Anger is usually regrettable, kindness rarely.

closed 2025-05-25 by digiexchris