jbruce12000/kiln-controller #208: errors reading Adafruit MAX31856 with S thermocouple

aaron4ce, 2025-04-02:
Hi, after two days of installation I finally ran kiln controller using Pi 3B+ and Adafruit MAX31856 while I use S type thermocouple. I have to admit that I lost a day not connecting MOSI as suggested with MAX31855. This is the difference while 31856 has to have this connection to provide at least thermocouple type setup. (obvious). I choose HW SPI and now I have something worst I can have and those are erratic readings!! It HAPPENS that the reading is correct. Nut it also happens that controller "hangs" on some older reading or reads erraticlally low or high value once and than goes back to the proper readings. Last time I checked the controller web access and worked good and when I started to run thermocouple test than the errratic readings begun. Than it hanged with some erratic reading... Please help me find this as reliable readings are must for that application. And frankly I've just bought a kiln with TC75 controller and I am terified: it is medieval. I was ofered to upgrade for TC95 which is less medieval but I vave to sell some gold to buy it. So please please help me because I believe in open software :)

this the output while testing:

pi@KILN:~/kiln-controller $ source venv/bin/activate
(venv) pi@KILN:~/kiln-controller $ ./test-thermocouple.py
Hardware SPI selected for reading thermocouple

board: RASPBERRY_PI_3B_PLUS
thermocouple: adafruit max31856
Degrees displayed in c

2025-04-02 12:14:18.972917 30.33C
2025-04-02 12:14:20.065077 30.27C
2025-04-02 12:14:21.178804 30.26C
2025-04-02 12:14:22.346985 30.33C
2025-04-02 12:14:23.515127 30.32C
2025-04-02 12:14:24.618175 30.33C
2025-04-02 12:14:25.733769 -452.00C
2025-04-02 12:14:26.837010 30.27C
2025-04-02 12:14:27.951555 30.27C
2025-04-02 12:14:29.021844 30.34C
2025-04-02 12:14:30.069653 30.27C
2025-04-02 12:14:31.151389 30.46C
2025-04-02 12:14:32.210513 30.38C
2025-04-02 12:14:33.269873 30.37C
2025-04-02 12:14:34.329373 30.38C
2025-04-02 12:14:35.378134 30.38C
2025-04-02 12:14:36.448217 30.39C
2025-04-02 12:14:37.498220 249.92C
2025-04-02 12:14:38.524060 249.92C
2025-04-02 12:14:39.594816 249.92C
2025-04-02 12:14:40.654538 249.92C
2025-04-02 12:14:41.703183 249.92C
2025-04-02 12:14:42.711739 0.00C
2025-04-02 12:14:43.782386 249.92C
2025-04-02 12:14:44.842076 249.92C
2025-04-02 12:14:45.903806 249.92C
2025-04-02 12:14:46.919036 249.92C
2025-04-02 12:14:47.978282 249.92C
2025-04-02 12:14:49.059416 -1556.00C
2025-04-02 12:14:50.140983 249.92C
2025-04-02 12:14:51.211565 249.92C
2025-04-02 12:14:52.270940 249.92C

aaron4ce, 2025-04-02:
After rewiring to software SPI and after reboot the web ui worked fine. I warmed the thermocouple with heatgun and measurements seemed to be accurate. And again after running tests everything went wrong and what is interesting after reboot readings started to "hang" once more. Here are the results of tests:

pi@KILN:~/kiln-controller $ source venv/bin/activate
(venv) pi@KILN:~/kiln-controller $ ./test-thermocouple.py
Software SPI selected for reading thermocouple
SPI configured as:

    config.spi_sclk = 18 BCM pin
    config.spi_mosi = 25 BCM pin
    config.spi_miso = 23 BCM pin
    config.spi_cs   = 24 BCM pin

board: RASPBERRY_PI_3B_PLUS
thermocouple: adafruit max31856
Degrees displayed in c

2025-04-02 14:38:05.239897 28.08C
2025-04-02 14:38:06.309900 27.98C
2025-04-02 14:38:07.314369 -2047.99C
2025-04-02 14:38:08.384294 27.91C
2025-04-02 14:38:09.454209 27.95C
2025-04-02 14:38:10.513184 27.90C
2025-04-02 14:38:11.583000 27.88C
2025-04-02 14:38:12.587515 256.00C
2025-04-02 14:38:13.657342 111.00C
2025-04-02 14:38:14.737987 27.73C
2025-04-02 14:38:15.807796 27.65C
2025-04-02 14:38:16.877657 27.69C
2025-04-02 14:38:17.947514 27.64C
2025-04-02 14:38:18.951938 0.00C
2025-04-02 14:38:20.021812 22.47C
2025-04-02 14:38:21.102877 27.52C
2025-04-02 14:38:22.173591 27.55C
2025-04-02 14:38:23.232542 27.48C
2025-04-02 14:38:24.302320 27.45C
2025-04-02 14:38:25.306756 2048.00C
2025-04-02 14:38:26.376742 0.00C
2025-04-02 14:38:27.446444 27.41C
2025-04-02 14:38:28.505429 27.33C
2025-04-02 14:38:29.575173 27.33C

BlakeCLewis, 2025-04-02:
All the issues like this that I experienced were due to thermocouple connections. Tighten them, pull on them, wiggle them.  And just curious, what extension wire are you using from thermocouple to  MAX31856?

rondoc, 2025-04-02:
Thanks good answer
 Also the wires from the tc to the controller need to.be matched to the tc.
We are talking about thousands of a volt here!!
Ronald.

Anger is usually regrettable, kindness rarely.

aaron4ce, 2025-04-02:
Thank you for your suggestions. I've checked the connections, I've tried other module too. I am pretty sure that those are ok. But maybe I am making other mistake? I run the test while kiln-controller is running. Well it is stopped, but the controller software is running from the boot. I've tested for a day now and all is working well unless I start test. btw. is there an option to stop the controller after it is activated after boot? SO those errors are present only during test procedure. Temperature readings during normal operation are accurate and stabile.

BlakeCLewis, 2025-04-02:
maybe:
sudo systemctl stop kiln-controller

BlakeCLewis, 2025-04-02:
2 processes using the same device on spi bus, will cause problems.

aaron4ce, 2025-04-03:
Thanks, that was the issue. So I rushed with running the service, I thought that tests are a bit more independent. But thats fine, everything is working nicely. I just want to arrange a new controller in such a way that I can plug into my kiln instead of TC75. But this should be simple now. Many thanks guys for help and assistance.

closed 2025-04-03 by aaron4ce