jbruce12000/kiln-controller #209: Is the partially controlled cooling possible?

aaron4ce, 2025-04-03:
Hi, I have a question while as I understand heating curve programming, but I am not sure is it possible to cool in partially controlled way? So for example from Tmax to T1 natural cooling, from T1 to T2 with 100 deg/hour down slope and than from T2 natural cooling to the end. If I program from Tmax to T1 in no time there will be "slower than programmed" but does the controller adjust the period from T1 to T2? Some glazes like controlled cooling. Kilns naturally cool in exponential way the the controlled cooling slope is limited.

rondoc, 2025-04-04:
Unless you have a cold air inlet the cooling rate will follow Newton's law
of cooling. You can use the kiln program to.make cooling that is slower
than the natural rate. I usually keep at cone 10 for about 15 minutes
before letting my kiln cool naturally. I have very small kiln that I do
control the cooling as naturally it cools too fast. All doable with the pi
control.

aaron4ce, 2025-04-05:
Thanks for response. I think I have to test the behaviour.

closed 2025-04-05 by aaron4ce