jbruce12000/kiln-controller #210: Additional SSR

stevo518, 2025-04-05:
I have been using the controller with great success, but I was wondering if someone could help me with a little modification I am wanting to do.

I have two elements currently being powered through 1 SSR, but I am wanting to have each element powered from its own SSR. I would like to program another pin to control this additional relay (mimicking the existing relay), but haven't been able to do this successfully as of yet. 

If someone is able to help me out it would be greatly appreciated.

Cheers

rondoc, 2025-04-05:
If you expecting to control each element differently then some new coding
is needed. Or you could just set up two pi controllers. If the elements are
going to be on and off together i.e. in sync then the single output from
the pi can goto a single transistor that can easily drive each SSR.

stevo518, 2025-04-05:
Thanks for your reply.

I realise I could switch both outputs with the one transistor but I was wanting to use a separate pin and transistor so that in the event that the transistor fails I would only have one SSR stuck on instead of two

Cheers

rondoc, 2025-04-05:
Don't see your logic in this . There are many paths to failure why chose
this one. Less is often best.
Ronald.

Anger is usually regrettable, kindness rarely.

BlakeCLewis, 2025-04-05:
I use a 18 pin DIP, uln2803a ($1.00), to switch the 12v for the relays.  It does 8 channels on 5v logic,  500ma per channel. I use 3 channels, 1 for each ring of the kiln. Advantages: smaller wire, smaller relays, easier cable management, and scalable.  I could add a 4th ring without changing any existing internal wiring, or scale down to 2 rings with same parts and pieces. I could also change out the software to software that uses zones, with a thermocouple/zone.

rondoc, 2025-04-05:
I suggest you use three TCS with three tc modules and an Arduino to control
each zone. Simple code using a 16 step power structure. That way the
heating would be very uniform. No need for PID either. You can easily add
a  mobile phone setup App and control using bluetooth .
Ronald.

Anger is usually regrettable, kindness rarely.

stevo518, 2025-04-06:
> Don't see your logic in this . There are many paths to failure why chose
> this one. Less is often best.
> Ronald.
> 
> Anger is usually regrettable, kindness rarely.
> […](#)

There are a few reasons for having two SSRs
1. Cost, I don't like the idea of the Fotek relays that many seem to be using so I opted for a different brand and it was cheaper to get two smaller relays rather than one larger one
2. Voltage drop, I am seeing a large amount of voltage drop at the elements using only one phase, using two phases will greatly reduce the voltage drop.
3. The added benefit that if one of the SSRs was to fail I would only gave one element stuck on, with only one element on I doubt I the Kiln would exceed its Max operating temperature. I do have another safeguard in place but I would prefer to reduce the likelihood of it having to operate as much as possible.

I am wanting both SSR to operate in sync, it made sense to me to have it switch from a separate output to reduce the chance of both elements being stuck on even further.

Sure simple is best but it is hardly adding any complexity. I figured this would be a very simple program change even for someone with no python experience like myself but unfortunately at this point I haven't been able to execute it.

Cheers

rondoc, 2025-04-06:
I am concerned that you are seeing a voltage drop . The SSR drop is about a
volt or two. Where is the drop? SSR s that can handle a 240 volt 10 kW kiln
aren't that expensive are they. I bought mine in 1987 so may be I'm
dreaming..
Ronald

rondoc, 2025-04-06:
I might add that chatgpt is very good at coding..just give it the details
and it will give you the code. You can send the code to it for a checking
or debugging.

stevo518, 2025-04-06:
The voltage drop is caused by the wiring on my property, using two relays means I can use two phases so therefore the current is halved.

Awesome, I'll give chatgpt a shot 👍

rondoc, 2025-04-06:
Seems you are using both legs of a 240 system . This means the current will
have to return via the neutral. Is there a reason why you don't just use
the 240 ?
Ronald

Anger is usually regrettable, kindness rarely.

stevo518, 2025-04-06:
I am using two phases and the neutral of a 400v system (230v).

stevo518, 2025-04-25:
Chat GPT was able to help with the coding, cheers

closed 2025-04-25 by stevo518