jbruce12000/kiln-controller #212: Unable to connect to server

jvswan, 2025-04-14:
I'm sure I'm missing something simple... I've installed Kiln Controller on my Raspberry Pi Zero 2W using the Read Me doc. I've been able to SSH into the Pi and I've updated the config.py and verified the listening port. I'm trying to pull up the interface to test and every time I get an error saying the browser can't connect to the server. I'm using the IP of the Pi and port 8081 (http://192.168.xx.xx:8081). 

Based not the Read Me, seems like I should be able to enter the IP and port and see the interface. Any suggestions? Perhaps I need more documentation or something. Are there any other sources for installation and configuration information out there?

TIA

BlakeCLewis, 2025-04-14:
> A few things you can do to troubleshoot
> check for the kiln process
> look at port usage
> hit the page from the pi w
>
> ssh to pi w
>
> make sure process is running with:
> 'ps aux |grep kiln-controller'
> That should return process info
>
> 'netstat -l |grep 8081'
> That looks at all of the listening ports, and should return a line for the
> process using 8081
>
> you can also hit the port with telnet and send a request:
> 'telnet 192.168.xxx.xxx 8081'
>
> then send:
>
> GET / HTTP/1.0
> Host:192.168.xxx.xxx
> <enter key>
> <enter key>
>
> that should get a text representation of the webpage

when address itself you can use locahost instead of 192.168.xx.xx
telnet localhost 8081

BlakeCLewis, 2025-04-14:
oh and to get out of telnet ctrl-] and then ctrl-d

keebie81, 2025-04-15:
What OS? I know on my mac chrome doesnt work reliably. Even though I have local network enabled for it, it wont work with local network adresses. I have to use the safari browser. So check your system settings that local network is allowed for browser you are using

rondoc, 2025-04-15:
I know the mobile connection works every time with an Android phone. I just
open the app locally and leave it open then I goto my.mobile enter the xxxx
8081 stuff and bingo it's there. At the same time I can see the pi info
tracking my mobile activity.
Since your mobile is an Apple may be it's blocking the remote pi server ?
Ron
Ps you can get a good android phone for under $30 on ebay...

Anger is usually regrettable, kindness rarely.

keebie81, 2025-04-15:
Safari works perfectly fine on desktop and mobile. Chrome used to, something in mac os or chrome changed a few months ago. Ive just learned to use safari

rondoc, 2025-04-16:
I don't know why but the link on my phone is
http://192.168.4.33:8081/picoreflow/index.html . I did not add the stuff
after the 8081/  it just appeared!
the 4.33 is unique to me i believe.
ron.

closed 2026-10-10 by jbruce12000