jbruce12000/kiln-controller #213: Kiln controller thinks that run is completed before scheduled temp

harry17111, 2025-05-07:
I have the controller set to 2266 in 450 min , but when it hits around 1850 F ish, it shuts off and says run completed. Any suggestions?

rondoc, 2025-05-07:
Send the code it will show why it occurred.

harry17111, 2025-05-07:
May 14 22:36:09 kiln python[350]: 2022-05-14 22:36:09,824 INFO oven: temp=1888.40, target=1888.00, error=-0.40, pid=54.33, p=-3.99, i=69.11, d=-10.79, heat_on=1.09, heat_off=0.91, run_time=27250, total_time=27335, time_left=84

rondoc, 2025-05-09:
does not show what happened at the shut down.. why is that ? Anyone?

harry17111, 2025-05-09:
Here is some more info-

- The max temperature that it was hitting (around 1850) is consistent in all runs I do now.
- I have deleted the entire program and reinstalled it (didn't work)
- 1850 F was the temperature of my initial bisque run , (the value is possibly not random) ; might be drawing from that somehow?
My next plan is to factory reset the raspberry pi, to see if it clears up something that shouldn't be there

rondoc, 2025-05-09:
There is more data that might help you. Can you do a data dump. I'm not a
coder so cant help sorry..BUT Chatgpt is great at reading  code and giving
back information.
This problem of suddenly shutting down is in the history...may be someone
can remember the outcome.
I usually blame wiring and the tc connections especially making sure the tc
is not grounded. The chip that does analysis of the tc is very sensitive to
a ground of the tc wired.
Ronald.

Anger is usually regrettable, kindness rarely.

keebie81, 2025-05-11:
You can duplicate this by unplugging the thermocouple while the program is running. So possibly the thermocouple is going bad or having interference at the higher temperature. Which is causing the program to think the thermocouple is no longer connected.

digiexchris, 2025-05-22:
troubleshooting tip: kiln-controller outputs logs to stdout/stderr, if it s setup to start on boot you can get the log via
`sudo service kiln-controller status`
Last time I had a shutdown without it ending normally I was able to get the error that caused the shutdown that way.

jbruce12000, 2025-06-02:
To troubleshoot this problem, I need to see the logs. There is a script called ```ziplogs``` in that will grab all the logs into an archive. then you can post them somewhere and link them here.

jbruce12000, 2025-06-03:
You can set [ignore_tc_too_many_errors](https://github.com/jbruce12000/kiln-controller/blob/a2b3071e4e55f47c20326563200da0b49d3c5bb8/config.py#L246C1-L246C26) = True in config.py. If this is a thermocouple problem, it will be ignored. You should really view the logs to see what the problem is though.

Nikodemuzz, 2025-07-22:
Hi all! I have a similar sounding issue as the original poster. I have completed the installation of the controller, and the initial tuning of the PID (or so I think at least). I had some issues during the installation with my thermocouples, which I documented in another thread here. They are solved now. The only other peculiarity was that at some point of the process I had to change the "Invert output" to True, because the SSR would turn on at startup, and when running the test_output.py it would work on reverse. Don't really know why that happened, but I don't think it has anything to do with the issue at hand.

Today I tried to run the first actual kiln program. To me it seems everything started fine, but maybe after half an hour or so the controller stopped and told me the run is completed. The temperature was 56C or so at the time. I started the program again, and got the same result in a few minutes. I ran the zip_logs script, but was unable to read anything from the resulting file. I will nevertheless attach the file here.

Any ideas what might be going on? Reading @jbruce12000's response above it might well be that there is some noise or errors in the TC reading. 

By the way, what is the significance of the rightmost indicator in the top right corner of the UI? The red "heat" marker? For me that one is constantly on.

[kiln.logs.gz](https://github.com/user-attachments/files/21367132/kiln.logs.gz)

gadgetmies, 2025-08-05:
@Nikodemuzz: It seems the zip you provided is empty. Perhaps you should try to reproduce the issue and run `journalctl -u kiln-controller` to see if that would return log entries.

closed 2026-10-10 by jbruce12000