jbruce12000/kiln-controller #215: Time remaining and timescale on graph seem to be adjusting to temp schedule, not actual time

pennltrex, 2025-05-20:
I've been running some tests. I'm currently running a test schedule, heating to 2250 over two hours, holding at 2250 for an hour, then cooling for an hour, over a total of 4 hours.

2 hours in, the temperature sensor graph matches the target temp graph, but the elapsed time is only about 1:30 and time remaining shows 2:27:47. Heat Rate has been around 1000f most of the run. 

While running the included "test-fast" schedule yesterday, which starts at 200f, the heat rate was 9999 for the first few minutes while it warmed up to 200f, so it isn't acting as if it is behind schedule. 

I haven't run a test to see how fast it gets to 2250 using the kiln sitter, but I would expect the heat rate to go to the max if it was struggling to heat at the rate requested. Am I misunderstanding how this is supposed to work?

[image]

rondoc, 2025-05-20:
There is a rev of the app that shows real time . Not sure if that is now
the latest app.  Also the original app shows time from a start and the
intervals are not the easiest to use. Imnsho.
The x axis would be nicest if it was divided into 15 minute intervals.
Ron.

Anger is usually regrettable, kindness rarely.

pennltrex, 2025-05-20:
@rondoc This version was downloaded just a few days ago, so it should be the latest.

Are you talking about a fork of this app?

rondoc, 2025-05-21:
I'm not sure but I remember a fork some time ago that changed the x axis to
be real time.

Anger is usually regrettable, kindness rarely.

jbruce12000, 2025-06-02:
See config.py setting [kiln_must_catch_up](https://github.com/jbruce12000/kiln-controller/blob/a2b3071e4e55f47c20326563200da0b49d3c5bb8/config.py#L193C1-L193C19).

Also see the /state endpoint which shows real-time debug status of the kiln.

closed 2026-10-10 by jbruce12000