jbruce12000/kiln-controller #219: Question about possibility of controlling two SSR units from the same GPIO

TinMaan, 2025-06-22:
I will be controlling an ancient smaller L&L kiln from the 60s ( 1960s ) as a test kiln. It runs on 240V/single phase - three wires ( two hot and neutral ). This little Oldy but Goldy fires a hot cone 9 in ten hours drawing 25 amps.  The rudimentary controls are three knobs for bottom, middle and top heating elements. The original idea was to replace those knobs with a 3 Phase SSR and be done with it. Keeping the SSR cool in that location turned to be nearly impossible and I am not keen on extending wires etc.

The cool idea is using two SSRs on the main cable and mounting the project box with proper cooling fan on the wall. Can I run two  Bc547 transistors from the same GPIO to power the two separate SSR?

I understand there are logic level converters that could be used for this case but I can't figure out how to use the reliably.

TinMaan, 2025-06-22:
> A simple solution would be to use just one SSR. Cooling with 25 amps
> flowing thru the SSR is no problem . The power lost in the SSR is only
> about 25 to 30 watts. A small fan will do that. Can you explain why you
> want multiple  SSRs?
> […](#)

I assumed I need two SSRs as there are two hot lines powering the kiln. I have read a few heated discussions on the internet on the subject to no end. Some people strongly advise to cut both hot lines with SSR on each hot leg and some people ( including pro electricians ) claiming it's un-necessary or over protectionist using two SSRs and one will do the job.

Honestly I don't know and I would prefer simple workflows as I am only a simple man. So, I am open to suggestions.

BlakeCLewis, 2025-06-22:
If you do not switch both hot wires, the elements stay energized and
touching them will shock you. You can get a 3 phase SSR and use 2 sets of
contacts.
If you decided to only switch one side, I would not load/unload without
disconnecting power.

I use 3 - 2 wire 35amp relays, one for each element (16 amp elements), then
3 sets of 10 awg wire to the kiln. If I switched with one big relay, the
wire would be 6 awg, and a 60 amp relay/ssr is more expensive. I took apart
some cheap 40amp SSRs, They had 15amp components.

I use a dip chip, unl2003a, 8 channels, to switch 12v with gpio pins. I
have a channel per relay and I use a few more channels for fans.
The MAX31856 and MAX31855 have an internal thermocouple that I use to
monitor controller temp, compare it to room temp and switch cooling fans in
the control box. I keep meaning to control the fans with pid and pwm. Right
now just off/on.

3 gpio pins, one for each relay allows individual control and it is
modular. I have a second kiln that has only 2 (16 amp) elements and I can
switch the controller to either kiln with 30 amp twist lock connectors.

TinMaan, 2025-06-22:
> If you do not switch both hot wires, the elements stay energized and
> touching them will shock you. ...
> […](#)
Yes, I perfectly understand that vital point. As a general practice of years, I never load or unload electric kilns powered. Also nobody else is allowed ( ... er crazy enough ) to enter my kiln room. Even then I have a sign stating that the main panel must be off before loading or unloading any of the kilns. That rule includes the newer kilns with modern Bartlett controllers  
Thank you for this important reminder.

TinMaan, 2025-06-29:
> I see no reason to switch both sides of a 240 system. Yes it's true if you
> touch ground and the non switched line you'll get 120 volts pass through
> you. However why would you risk that. Have a full breaker and disconnect
> everything when loading etc. Just switch one a side it's much easier. BTW.
> as an EE  I wouldn't trust the SSR to provide " touch free" isolation. They
> have leakage current that might be felt.
> REMEMBER KISS...
> […](#)

Here is an update. The electrician finally dropped in. After a bit of pondering on my old hand drawn schematics that oroginally was done and issued by L&L ( of the bygone era ) and then a bit of organizing a jungle of wires creeping everywhere indie the " OEM control box ", he said : " We have two options. A- getting rid of the control knobs and ganging the hot wires into two and then controlling one of the two with the one SSR discussed here. ... or ... B- using two separate SSRs on existing mains before entering the kiln ( on the wall ) and with that we can keeping the control knobs for fine tuning zone control."
What would you do? 

Here is a copy of the old schematics wiring of the little kiln:

[H1500 L&L DynaKiln wiring diagram - A-103X.pdf](https://github.com/user-attachments/files/20968996/H1500.L.L.DynaKiln.wiring.diagram.-.A-103X.pdf)

rondoc, 2025-06-29:
I have 24 volt 3 phase  contactors unused and good for at least 25A you
can have for shipping cost . You could use them for safety or whatever.
I would have one SSR controlled by the pi kiln controller. And if the
existing switches are functional leave them in the system. The SSR will
switch the 240 incoming  to whatever setup you have with existing switches
. It doesn't care! Then you can have best of both worlds. The local
switches will allow for balancing the temps and the pi will try and keep
the temp profile you've programmed. You will.need to.set up the PID logic
so they.match your kiln. Don't worry if the pi doesn't track at start up.
It should follow within a degree after about 300F.

This is the simplest solution i think. Don't try multiple SSRs it's not
necessary.  I would have a main breaker near by or easily accessible. It
seems that a 50 amp.breaker would be fine if not a bit overkill.

Ronald.

Anger is usually regrettable, kindness rarely.

TinMaan, 2025-06-30:
> I have 24 volt 3 phase  contactors unused and good for at least 25A you
> can have for shipping cost . You could use them for safety or whatever.
> I would have one SSR controlled by the pi kiln [...]

I also prefer to keep things as simple as possible. I assumed I was able to place the single SSR on one of " hot " legs of 240V power cable ( three wires 50 amp ). They had wired the top zone of this little kiln to one hot wire ( L2 ) and the middle+bottom of the kiln is wired to the other hot wire ( L1 ). With this configuration, I am not sure how to control all the power to all the zones with only one SSR. 
Could you take a look at the attached wiring diagram of the kiln and suggest how to proceed with the idea of controlling the kiln with only one SSR please?

[H1500 L&L DynaKiln wiring diagram-worked - A-103X copy.pdf](https://github.com/user-attachments/files/20971097/H1500.L.L.DynaKiln.wiring.diagram-worked.-.A-103X.copy.pdf)

rondoc, 2025-06-30:
I'd like to help but I don't know what the switches do. I can guess but
that's all. WITH NO POWER CONNECTED you could take a VOM and find out what
the switches do. To do this make a table. Position 1 : A connects to what
B connects to what
Same for all three positions.
It seems that the original system ran I'm some positions with elements on
120 volts and the max went to 240.
I wish I was close by as the challenge is not hard but trying
to.help.remotely is very difficult.
The simplest solution would be to use all elements together going to.one
SSR powered by 240. No.connection the neutral. This would be like a
domestic 240 oven. This won't help to balance the temperatures .
Another suggest is use the pi only for monitoring and control kiln
manually. It works well for that.
Sorry not to.be more help.
Ronald.

TinMaan, 2025-06-30:
> I'd like to help but I don't know what the switches do. I can guess but
> that's all. WITH NO POWER CONNECTED you could take a VOM and find out what
> the switches do. To do this make a table. Position [...]

Thank you for taking the time to reply. In fact your suggestion for ignoring the switches and " binding " all the heating elements to be powered as an electric oven is similar to what the electrician is calling the plan B. This kiln is small enough to fire uniform enough that way. 
I have contacted L&L tech also. Occasionally they help with user friendly cases of maintenance ( of their own machines ). They  might have a suggestions.

rondoc, 2025-06-30:
Last idea when you have all the elements connected as one you or the
electrician can determine the current load.
Then choosing a single SSR will be better. The reason for using a SSR is
that the pi system switches often and a contactor would be worn out quickly.
SSRs dont worry about that. Best and let us know how it worked out.  One of
the best things about the pi control is that you can monitor the kiln from
your phone
. I often start late in the day and check from my bed during the night.
ronald.

TinMaan, 2025-06-30:
> The schematic shows that each element is 120V. It is a bit weird, it is not
> a balanced load,  L1 has 2x the load as L2.
> Each knob/switch gets a Neutral and a Hot, either L1 or L2. Top switch, L2
> and N (120V), and the bottom 2 switches, L1 and N (120V). No elements have
> both L1 and L2 (240V).
> 
> The way to rewire and run it on 240v is to run each pair of elements in
> series as defined below. This will create 120V drop per element, which is
> what the switched high setting would have been.
> Get rid of the switches. I have found that one of the issues with old kilns
> is all the electrical contacts are a bit corroded, and the resistance adds
> up and people think it is an element problem. Have your electrical buddy
> turn the kiln on high and measure the voltage of each of the 6 elements. I
> am pretty sure it will be 120V. If you want to control zones, 3
> thermocouples and 3 SSR/relays, and do it in software.
> 
> If 120V/element, then this will work:
> L1---element1----element2----L2
> L1---element3----element4----L2
> L1---element5----element6----L2
> 
> If you choose to do this, use this math: (the kilns rated Watts)/240 =
> Amps. Then (Amps x 1.25) is the minimum breaker size and is what the wire
> should be sized for.
> 25amps * 1.25 = 31.25amp, you can fudge it a bit and use 30 amp breaker and
> [#10](https://github.com/jbruce12000/kiln-controller/issues/10) awg wires, or if running over 20' then [#8](https://github.com/jbruce12000/kiln-controller/issues/8) awg wire on a 40amp breaker.
> 45amps * 1.25 = 52.25,  so a 60 amp breaker, [#6](https://github.com/jbruce12000/kiln-controller/issues/6) awg and if you are running
> wire over 20' use [#4](https://github.com/jbruce12000/kiln-controller/issues/4) awg
>  I say calculate the amps, because if it is running 120v elements and you
> convert to 240v, the amps may be different, but the Watts will be the same.
> 
> I am pretty sure the knobs work the way they did on my Skutts: My skutts
> use 240V per element.
> 
> All of these are 120V
> Low:  1 element of each pair is on
>     (L2---element1 ---N)
>     (L1---element3 ---N)
>     (L1---element5 ---N)
> Med: all elements are on, but run in series of 2, each element drops 60V
>     (L2---element1---element2---N)
>     (L1---element3---element4---N)
>     (L1---element5---element6---N)
> High: all 6 element are on in parallel
>    (L2---element1---N)
>    (L2---element2---N)
>    (L1---element3---N)
>    (L1---element4---N)
>    (L1---element5---N)
>    (L1---element6---N)
> 
> At 240V: you would wire it like the med setting: each element drops 120V.
>     (L1---element1---element2---L2)
>     (L1---element3---element4---L2)
>     (L1---element5---element6---L2)
> And it would output like the high 120v setting.
> 
> 
> I would switch it with a SSRs. These 2 SSR configurations will switch both
> L1 and L2 at the same time:
>    Double Pole, Double Throw (DPDT), which has both NO and NC, but you only
> use the NO contacts (Normally Open);
>    Double Pole, Single Throw, which I have only seen Normally Open (NO).
> I used 3 relays (about $20 each), one for each pair of elements. I use
> relays, but my switching frequency is once every 30 seconds. At high
> switching rates SSRs will be a better choice.
> I mount everything in an external box about 1 meter away from the kiln. Put
> heatsinks on SSRs and move some air over them.
> If you use 1 SSR, use one that is rated at about 1.5x the Amps of the kiln
> calculated above.
> 
> Use high temp appliance wire, glass fibre insulated, also appliance crimp
> terminals. They are plated steel and take some muscle to crimp. The crimper
> is different than the one used for normal terminal crimps.
> Amazon has both.

Thank you for this thorough and complete " How to ". I learned a lot. 
Yes you are correct about those switching knobs. One of them has lost one step and another one intermittently disconnects. The electrician said, he might be able to replace them with new ones.
Good point about the heating elements being 120V. With that, they have to be powered separately on each hot leg ( two of the in parallel )
I found a similar case in archives of Ceramic Daily. Somebody had tried to convert his older SKUTT kiln ( with kiln sitter and switching kmobs ) to run on cheap PID controllers - SSR combined. Other members cautioned about the usual concerns about the use of SSRs and the need for additional safety contactor etc. But generally they agreed with the concept of wiring and idea. Here is the OP wiring diagram:

[Kiln Controller - DIY.pdf](https://github.com/user-attachments/files/20972850/Kiln.Controller.-.DIY.pdf)

rondoc, 2025-06-30:
I think the simple run all levels at 240 with one SSR and use the pi
controller is the easiest. It's simple cheap and reliable.  Other ideas are
great but this is an old kiln and KISS really applies unless you have time
and money..if you have the later buy a new kiln. The old ones are not
efficient as insulation is far better now so operational cost will be
lower.

rondoc, 2025-06-30:
> Yes, that's how I intended to do this. But the in person inspection of the
> kiln by an electrician and understanding the wiring is forcing me to take
> the alternative rout:
>    - The top zone of the kiln is wired to one hot leg of 240V and the
>    middle combined with bottom zone is wired to the other hot leg.
>    - The heating elements are all 120V individually.
>    - In order to fire and control all the heating elements with one SSR I
>    will have to change over the heating elements to 240V type.
>    At this point it seams a lot more convenient to control each 120V leg
>    of the three-wire 240V separately and keep the switching knobs in place for
>    the fine tuning the zones with trial and error. I also assume two separate
>    sets of controllers will have a better chance to fine tune the PID and keep
>    the heating more uniformly and that is just the bonus point.
>    Unless I am missing something as usual.

You just connect the 120 elements in series 2 together make a 240 volt.one
You will then have 3 240 volt elements
Top  middle and bottom. You then connect these 3 in parallel making ONE 240
element equivalent.  This will.have 2 ends one will.go to one of the 240
lines the other to the SSR and then to the other 240 v. Leg.
Got it?

rondoc, 2025-06-30:
> Yes, yes it sounds quite simple to do. I will sketch a simple diagram for
> sanity check with you before getting it done.
> BTW, in this configuration what will happen to the neutral wire of 240V
> cable ( three wire 240V )?

No connection to neutral.unless you want a 120 supply for pi controller set
up.

TinMaan, 2025-06-30:
> No connection to neutral.unless you want a 120 supply for pi controller set
> up.
> […](#)

here is the simplistic schematics digram for creating the three sets of 240v elements in parallel:

[image]

BlakeCLewis, 2025-07-01:
A third wire for a case ground is needed. If the controller is using 120v
for a pi power supply, then you need a 4th wire for neutral, or a second
circuit providing 120v.

BlakeCLewis, 2025-07-01:
Otherwise the schematic looks correct

TinMaan, 2025-07-01:
> A third wire for a case ground is needed. If the controller is using 120v
> for a pi power supply, then you need a 4th wire for neutral, or a second
> circuit providing 120v.
> […](#)

The control box(s) will be powered with 120v wall plug-in ( through a small UPS ) as we get brownouts and short ( merely seconds ) blackouts in our region.
I have been thinking about grounding the kiln wiring. Can I use the neutral wire of my three-wire 240 cable for grounding the kiln?

TinMaan, 2025-07-01:
> Otherwise the schematic looks correct
> […](#)

It's great to know. I really couldn't imagined this project would be this involved. Honestly, I think a Bartlett V6 at about $200 gets the job done. Buy, hey - I learned a lot and that makes the whole troubles worthwhile.  Thank you for all the help.

rondoc, 2025-07-01:
When two 120 volt coils are in series to make a 240 volt one each of the
coil gets 120 volts . The connection between the two needs to be very
solid. I don't know what wires some out of the kiln but making good
connections with hot ( temperature) wires is not trivial. It's best if
connections are made where it's cooler..
Ronald.

rondoc, 2025-07-01:
You can use the neutral wire as most ovens do that.  It's better to use a
ground wire that doesn't get used to carry any power. I have done it both
ways and although I am an EE I'm not an electrician who has to live with
building codes!

TinMaan, 2025-07-01:
> In electronics they call it voltage drop. With 2 elements (the elements
> match) in series, each will drop half the voltage. For 240V, if you measure
> across one of the elements, the voltage would measure 120V.
> 
> The way it was wired, on either medium or low (can't remember which),  2
> elements are run in series. With the 120V setup, measure across 1 of the
> elements, and it will be 60V.
> […](#)

... interesting! 
I never understood the reasoning behind the " voltage " classification of commercial ceramic kiln heating elements. This is another subject I have start learning because sooner or later I will need to replace the coils in this kiln. L&L does not make or distribute the elements for this old kiln anymore. I have to make the coils the DIY method. There are a few Youtube videos on the subjects.

rondoc, 2025-07-01:
Re making elements: get Kanthal wire and wind your own. It's pretty easy.
The gage of the wire  diameter and length of the wire is all calculable.
Someone can give it to.you. if not chatai  can give you the answer in
seconds!

TinMaan, 2025-07-01:
> You can use the neutral wire as most ovens do that.  It's better to use a
> ground wire that doesn't get used to carry any power. I have done it both
> ways and although I am an EE I'm not an electrician who has to live with
> building codes!
> […](#)

I get the electrician to take care of that inside the sub-panel.

TinMaan, 2025-07-01:
> Re making elements: get Kanthal wire and wind your own. It's pretty easy.
> The gage of the wire  diameter and length of the wire is all calculable.
> Someone can give it to.you. if not chatai  can give you the answer in
> seconds!
> […](#)

Where would you recommend buying bulk Kanthal wires. I've seen Youtubers use these massive rollers of Kanthal that supposed to bring the cost down.

rondoc, 2025-07-01:
Amazon is one source. Search the net if they don t have the gage you need..
KANTHAL A1 is fine upto cone 10. I would suggest you work.with cone 5 clay
and glazes they less are expensive and use about 40% less power to fire.

Anger is usually regrettable, kindness rarely.

BlakeCLewis, 2025-07-01:
DIY elements:
  I don't remember the numbers, these are just examples
  What I found when I looked into winding my own was I needed like 42' for
a single element. The rolls of wire I could find were like 120', so I could
only get 2 elements out of a roll, with 36' unused.

There are a few companies that make elements to order. They may have specs
for your kiln, or you give them spec on the kiln and they figure it out.
  www.theceramicshop.com has L&L parts

Element Voltages:
  My elements are designed to put out 11520W at 240V, but at 208V they only
put out 8652W.  They will work, but probably not get the kiln to temp.
  If you have elements designed for 208V, you may be able to use them at
240V and will have higher output, but they will not last very long.

You talked about using the knobs to balance the kiln. What Skutt did in the
past is make 2 different elements for my kiln. The top and bottom elements
ran hotter than the 4 in the middle to compensate for the heat loss due to
the surface area of the floor and lid.  A smaller wire diameter that heats
up faster. I think  they did away with that when they went with zone
control.

TinMaan, 2026-04-07:
>
> A better use of a second relay is to create a fail safe so the kiln doesn't melt down should the relay fail closed. Switch L1 and L2 concurrently using the same output from your controller but using dedicated SSRs on each leg. If one relay fails closed this will prevent your kiln from firing until failure.

Hi Scott, Could you elaborate on how to wire L1,L2 ( and two SSRs ) for redundancy and more importantly how to reconfigure the code to send signal to each SSR in correct sequences Please?

Villagepotter, 2026-04-08:
Scott Chatenever
Just as Blake explained.
L1-e1-e2-L2L1-e3-e4-L2L1-e5-e6-L2
Functionally, it's adequate to switch everything on and off together if you can source a big enough relay. Either a single relay switching L1, or an additional redundant relay for L2.
The only reason the original kiln design had elements separated into three groups was because the current limitation of those infinite switches required it. 
Did you complete this build? I'm confident with the electrical part, but know nothing about RPi and will be leaning on my programmer buddy.
Chatenever Ceramicshttps://linktr.ee/villagepotter

rondoc, 2026-04-09:
> On second thought, this isn't a great safety strategy because you wouldn't
> know if one ssr failed closed, and if both failed you'd be right back where
> you started.
> Better if you group the element pairs arbitrarily but evenly as possible
> into two circuits, each with its own relay (again both switched with a
> single output). If one fails open or closed the controller would see the
> temperature misbehaving and alert you. And failing that, the kiln would not
> over fire with only half the elements powered.
> That's basically how new kilns are built - no global safety, just relying
> on it being too underpowered to be a problem if one relay fails, and
> banking on the unlikelyhood of two failing simultaneously.
> You could use smaller SSRs too.
> Ron, I'd love to get coffee next time I'm in SB.
> Scott Chatenever
>
> Chatenever Ceramicshttps://linktr.ee/villagepotter

Scott.. contact me outside this grp. Rdoctors at gmail.

Anger is usually regrettable, kindness rarely.