jbruce12000/kiln-controller #223: Cannot get temperature to read correctly

kveldssanger, 2025-07-06:
Hello All!
I have been struggling with getting a correct read on my thermocouple. I am using a MAX31855 with a Pi Zero2W with Debian Bookworm installed using the Raspberry Pi Imager. Due to the common errors from some MAX 31855 boards I have tested my MAX31855 with an arduino and have confirmed that (with a small capacitor hooked up to the thermocouple like the forum suggested) it is reading the temperature fine on the arduino. I have checked this with two type K thermocouples and 2 separate kilns.

However, when I run it on the pie it starts at the seemingly correct temperature but then quickly lags far behind the actual temperature. For example, when I run the test-output script my my infared thermometer and another basic temperature reader tells me far different temperatures. When the temperature of the kiln got to 400degrees (verified by my other two deviced) the TC in the kiln connected to the pi still said it was 250degrees. It kept going up, it felt like some kind of delay because even when I turned the power to the kiln off the temp off the TC would keep rising.

I have gone through so many steps trying to fix this, even getting another MAX board and another Pi, reloading the OS many times, trying new TC wire, lengths and positions, testing entire system on another kiln, using a different TC, grounding everything to the kiln itself and scoring this forum for every possible solution I can try to correct this, to no avail.

Any guidance would be most helpful, I am at a loss.

Thanks,
Chelsea

[image]
[image]

rondoc, 2025-07-06:
I have the 55 chip from a good us source and I get tracking to cone 10.
Either you are using a fake board or you have a partially grounded tc.
I parallel.the tc to a Fluke and tracking us within a degree to cone 10.

rondoc, 2025-07-06:
It is possible you have severe rf interference but I can't think from what.
Old brushed motor fan or ???

kveldssanger, 2025-07-07:
I have no doubt that I this setup has the capability of working as yours is and I look forward to the day that it does haha Its just something is wrong and causing this issue, I have yet to know what that is but I am determined to find it!
Cant think of anything around the area of my studio that would be causing severe rf interference but I will look more into it. I have checked the TC for grounding and have confirmed that it is not grounded.

Im going to go ahead and order a MAX31855 off of Adafruit when they come back in stock since I know that board will be reliable. Ill let you know what I find when I get it in!

In the meantime if anyone knows of any other tests I can try, I'm open to anything.

Thanks for the feedback, more soon.

rondoc, 2025-07-07:
Look very very carefully at the pins you used on the pi. It's easy to be
wrong. Especially the ground pins. Also put a Fluke or other quality tc
reader across the tc . It will be in parallel with the pi wires. Also.make
sure the tc wires are oriented correctly.
It's probably something simple. To run my set up I open the pi and then
open kiln controller  if you have the app on your phone you then open the
controller.
Then it runs a short time and appears active on the phone. I use a
dedicated android phone for this. You the select the profile and go to
start.
Its mostly on the site.
Ronald .

Anger is usually regrettable, kindness rarely.

kveldssanger, 2025-12-23:
Hello again,
I finally got a proper adafruit Max 31856 to replace the questionable Max 31855 board/s I had before.

Since Ive been reading that the Debian Trixie OS from Rasp Pi imager has been causing others problems I decided to reinstall everything with the 32-bit Debian Bookworm OS and start the process fresh.

After the fresh image, I followed the instructions in the read me exactly as I always have, everything installed fine.
I checked my pins and connections on my pi and breadboard, checked the config.py file. 
Everything is set correctly with a capacitor across the thermocouple connection as described and recommended.

I run the ./test-thermocouple.py and all I get is 32 degrees.

I run the ./gpioreadall.py and this is what I get:

[image]

First problem seems like the pins are not set correctly. 
In an effort not to mess this up myself by cobbling together my own solutions,
can anyone please verify that I am correct in assuming the pins are not set correctly? 
And secondly, can anyone help me understand the process of how to change them?

I have tried to follow the troubleshooting.md document which explains how to do that but it never works and alway gives me this error:
bash: gpio: command not found

Also, just a side note, I have tested the Max 31856 with an arduino uno and it senses and measures temperature perfectly fine so the Max 31856 board is working as it should

Any help would be greatly appreciated!

Thank you!
Chelsea

kveldssanger, 2025-12-24:
Thanks for the reply, here is all that info.
And yes I did solder the header pins myself and yes they are firm and shiny. I also tested continuity with a multimeter.

import logging
import os
from digitalio import DigitalInOut
import busio

########################################################################
#
#   General options

### Logging
log_level = logging.INFO
log_format = '%(asctime)s %(levelname)s %(name)s: %(message)s'

### Server
listening_port = 8081

########################################################################
# Cost Information
#
# This is used to calculate a cost estimate before a run. It's also used
# to produce the actual cost during a run. My kiln has three
# elements that when my switches are set to high, consume 9460 watts.
kwh_rate        = 0.1319  # cost per kilowatt hour per currency_type to calculate cost to run job
kw_elements     = 9.460 # if the kiln elements are on, the wattage in kilowatts
currency_type   = "$"   # Currency Symbol to show when calculating cost to run job

########################################################################
#
# Hardware Setup (uses BCM Pin Numbering)
#
# kiln-controller.py uses SPI interface from the blinka library to read
# temperature data from the adafruit-31855 or adafruit-31856.
# Blinka supports many different boards. I've only tested raspberry pi.
#
# First you must decide whether to use hardware spi or software spi.
# 
# Hardware SPI
#
# - faster
# - requires 3 specific GPIO pins be used on rpis
# - no pins are listed in this config file 
# 
# Software SPI
#
# - slower (which will not matter for reading a thermocouple
# - can use any GPIO pins
# - pins must be specified in this config file

#######################################
# SPI pins if you choose Hardware SPI #
#######################################
# On the raspberry pi, you MUST use predefined
# pins for HW SPI. In the case of the adafruit-31855, only 3 pins are used:
#
#    SPI0_SCLK = BCM pin 11 = CLK on the adafruit-31855
#    SPI0_MOSI = BCM pin 10 = not connected
#    SPI0_MISO = BCM pin 9  = D0 on the adafruit-31855
#   
# plus a GPIO output to connect to CS. You can use any GPIO pin you want.
# I chose gpio pin 5:
#
#    GPIO5    = BCM pin 5   = CS on the adafruit-31855
#
# Note that NO pins are configured in this file for hardware spi

#######################################
# SPI pins if you choose software spi #
[... 196 lines cut ...]
# See https://github.com/jbruce12000/kiln-profiles
kiln_profiles_directory = os.path.abspath(os.path.join(os.path.dirname( __file__ ),"storage", "profiles")) 
#kiln_profiles_directory = os.path.abspath(os.path.join(os.path.dirname( __file__ ),'..','kiln-profiles','pottery')) 

########################################################################
# low temperature throttling of elements
# kiln elements have lots of power and tend to drastically overshoot
# at low temperatures. When under the set point and outside the PID
# control window and below throttle_below_temp, only throttle_percent
# of the elements are used max.
# To prevent throttling, set throttle_percent to 100.
throttle_below_temp = 300
throttle_percent = 20

**Images of my connections:**

[image]

[image]

[image]

[image]

Thanks so much for the help!

kveldssanger, 2025-12-26:
Wow! It's a Christmas miracle! It worked! 
For the first time in over half a year and countless hours.. 
4 different Max boards, 2 pis and probably 10+ OS reinstalls.... 
I can read the temp correctly and see the kiln controller interface.  
Incredible.

This is the only thing I got this Christmas and oh man am I grateful. What a gift!

To you Steele, Ronald and the whole community of people who contribute to this project, who kindly and patiently help artists like me who don't really know what their doing with Pis and the like...
Thank you all so much, you are appreciated!

Tomorrow I will run the tuning setup and hopefully that goes smoothly.
Ill let you know how it goes!
More soon. Fingers crossed.

Merry Christmas!

kveldssanger, 2025-12-26:
Since I work with both ceramic and glass, the closest I can get to an accurate tuning the better. 
If you have any advice or cheat sheet for this process, please do share. :) 

Regardless I will run a non-critical bisque and compare the profile lines to get me started.
More soon.

Thank you!

rondoc, 2025-12-26:
Anger is usually regrettable, kindness rarely.

kveldssanger, 2025-12-26:
Ron was there suppose to be a document of something attached to your last post?
I dont see anything specific.

Going to run the tuner now to see what I get before I do anything else.

rondoc, 2025-12-27:
> Ok, ran the tuner and almost immediately got some errors.
> Here are some screenshots of the errors I received.
> Screenshot.2025-12-26.at.1.36.23.PM.png (view on web)
> <https://github.com/user-attachments/assets/14789882-193b-4ca8-bc15-d49450516d66> Screenshot.2025-12-26.at.1.39.28.PM.png
> (view on web)
>
> Let me know if you guys have any ideas why this might be happening.
>
> Thank you!

It's attached.

Anger is usually regrettable, kindness rarely.

rondoc, 2025-12-27:
From your screen shot it looks like the TC is not connected or not being
read

Anger is usually regrettable, kindness rarely.

kveldssanger, 2025-12-27:
Im not sure if it just isnt loading or something but I dont see any attachments on your posts.
Is there something I need to do to access it? Or am I looking in the wrong place?
This is all I see:

[image]

And yes it does seem like the TC is reading like it is not connected and/or not being read.
But it is connected just fine and if I run test-thermocouple.py it will read the temp without error for long lengths of time. Would there be anything particular that would interfere with the TC while running the tuner?

Thanks!

steeleh, 2025-12-29:
As an example attached are a couple of shots of a brand new install on Pi Zero W with no tuning. Slightly wobbly for first few degrees but after that pretty much spot on.

Steele

[image]

[image]

AndrassGray, 2026-01-05:
These breadboards are notoriously known for their unreliable connections. And these Dupont cables are only slightly more reliable.

I am doing a bit of electronics myself, and I use them as well, but whenever a circuit doesn't work, I wiggle the cables and get my Ohm-meter to check connections.

rondoc, 2026-01-05:
I agree you cannot use that setup for a reliable system. Soldered connections are reliable. Ronald Sent from my Galaxy

rondoc, 2026-01-11:
> Well done for getting your controller behaving itself. Great set up once working properly.
> I am sure Ron is better placed to give you an explanation of "Heat Rate".
> Steele

Actually I'm.not the right one to ask. I have little idea of the software. It would be great if someone would go through it and comment in. PlainEnglish what the code was there for. Sent from my Galaxy

TinMaan, 2026-01-11:
> ... can you explain what the "Heat Rate" means exactly?

if you are referring to GUI ( web interface ) of the Kiln-Controller " Heat Rate ", it indicates the rate of firing also known as " slope ". That is a number indicating how fast you are firing to arrive at your next target temperature. I am sure you knew this, but the variable referred to differently around the world.
Noteworthy for new users is that the average presented " heat Rate " in GUI is short range calculated and must be reconsidered with experience for sanity check. I hope this helped.

rondoc, 2026-01-16:
> Hello everyone!
> Thanks for the answer about Heat Rate, that makes sense and made far more sense on this last test fire as the new schedule I set was longer and the heat rate correlated to degrees per hour the killn was increasing at. The other test schedules I had didn't work out that way.
> I had to do a few test firings over the past days to verify what was going on before I reported anything.
> Good news:
> Looks like the Pi Controller is working!
> It runs great, tracks the temp very well, follows the line on the schedule and seem to be doing its job.
> Bad news: My kiln cant seem to get past 1100F. It goes up to 1000 with a 300f/hr rate very well then once it gets to about 1000-1100 I just sort of hangs there for hours and doesn't move. Seems like the kiln itself is struggling to heat up.
> A little backstory, I was given three old fully manual/kiln sitter kilns from a friend of a friend and have been on a mission to get this PiController going so that I can use them all without spending all kinds of money on fancy controllers. They are used but they are in pretty decent shape outside of the manual controllers not working and a few chips here and there.
> First thought is that the elements aren't working as efficiently as they should. I did test them with my ohm meter and they all measured about 15ohms each which seems to be within spec for the Skutt 818kiln but in a few areas the elements coil is uneven and a bit stretched. Another thing to note is that the elements don't even start glowing until maybe 800F, I read this can be normal with some kilns but it seems odd to me as all the kilns I have used have elements that get red quickly after startup.
> [... cut ...]
> Tomorrow Im going to remove the 3rd ring of the kiln so it is smaller and maybe Ill get better results with that.
> 2200F seems very far away from 1100F with the way it was going so I am not too optimistic that it will get to the peak temp that I want but worth a shot, right?
> If you have any thoughts or anything else I should try/look into, let me know.

I checked with the skutt kiln. It is rated at 6.5 kw and at 240 voltages that isc6500/240 amps or 27Amos. The resistance is 8.8 ohms. ( 240/27). So if you measuring 15 0hms then the power would be 240 * 240/ 15. That is only about  3.5 kw. Not enough to heat past the 1100 degrees F. So check wiring of elements. Should 8.8 ohms. Sent from my Galaxy

kveldssanger, 2026-01-18:
So this is the one you would recommend?
https://www.mouser.com/ProductDetail/IDEC/RSSDN-75A?qs=LW8h5wBZGrFPCFpb%2FXPL6w%3D%3D

What other cheaper ones would you suggest that you have used successfully? Just wondering.

rondoc, 2026-01-18:
Plenty on ebay with similar ratings. Problem. Is many are not what they say or print.. So bottom line is buy from. MOUSER OR DIGIKEY. Pay more but a good one will outlast the kiln. Fan makes a difference and they are cheap. If the coils need replacing I can help with advice. The DIY coils are as good as the ones you buy. It is vital that the element ends are clamped very tightly. If they get even a bit lose you can get racing and quick damage. I use heavy copper washers. Ronald. Sent from my Galaxy

https://www.mouser.com/ProductDetail/IDEC/RSSDN-75A?qs=LW8h5wBZGrFPCFpb%2FXPL6w%3D%3D

rondoc, 2026-01-18:
The SSR if they fail usually go to a dead short!. Therefore it is prudent to have a safety switch in the form of a mechanical relay that cuts the power. A big clunky one is fine.. It can be switched to.open, from the pi via a transistor. If you watch your kiln when it gets near max temp. Then you can kill the breaker. I go to bed so I  am usually asleep then.. There are "kiln sitters" that do this.This conversation is outside the pi chats so if you want to continue I suggest my email is rdoctors with the usual Gmail. Ronald. Sent from my Galaxy

https://www.mouser.com/ProductDetail/IDEC/RSSDN-75A?qs=LW8h5wBZGrFPCFpb%2FXPL6w%3D%3D

TinMaan, 2026-01-18:
> Hello Ron, I'm thinking this relay which is well over what I actually need:
> 
> Schneider Electric - https://www.mouser.com/ProductDetail/Schneider-Electric/SSP1A175BDT?qs=GcQsFBwc5eX2td9xJwqVng%3D%3D
> 
> And here are some other options I found but they would be rated with volts and amps that are close to my hookup (which is 220V and 50amp) so I'm not sure if they qualify if the SSR needs to be well above actual as you mentioned. But let me know what you would suggest.
> 
> Crydom - https://www.mouser.com/ProductDetail/Crydom/D2450-10?qs=KoN42VEC6245W8CR2gbibg%3D%3D
> 
> Schneider Electric - https://www.mouser.com/ProductDetail/Schneider-Electric/SSP1A150BDT?qs=GcQsFBwc5eUEEb47mxVEGQ%3D%3D
> 
> IDEC - https://www.mouser.com/ProductDetail/IDEC/RSSDN-50A?qs=FESYatJ8odIDQ%252B8X6xUsQQ%3D%3D
> 
> Thanks for the help! Chelsea

Sorry for jumping in. _**Do make sure to double check and triple check the polarity of your low voltage side of the SSR**_. When you made sure of it, do it again. Many SSR including the expensive ones DO NOT HAVE PROTECTION against reversed polarity. Those will catch fire immediately when energizing the high voltage side. The smoky interior walls of my controller can tell you stories about back to back expensive incidents as the result of making assumptions about the DC side of SSR.

kveldssanger, 2026-01-21:
Messages received from all.
Thanks for the feedback! 
When I get the new ssr and get everything hooked up and triple checked in all ways... Ill let you know what happens.
Best,
Chelsea