jbruce12000/kiln-controller #224: Is my question about linearization of max31855 removed?

TinMaan, 2025-07-06:
I apologize if I broke any rules. I need to know what rule was it so I don't repeat.

jbruce12000, 2025-07-07:
I closed the issue because I answered the question. I'll let you close this issue, but here is my answer to the last issue:

The linearization you're referring to is called NIST linearization. It happens [here](https://github.com/adafruit/Adafruit_CircuitPython_MAX31855/blob/8cd15a76e4bd8a1bc8ca690b963e6bd1b156027e/adafruit_max31855.py#L116) in the code for the max-31855.

A few years ago, when I used my own library for this, I handled the NIST linearization. Glad to hand that off to someone else.

TinMaan, 2025-07-07:
Thank you very much for clarification and all the work you've done. It's reassuring to know this issue had been addressed for a person( me ) that cannot read code.
Please close this thread as you see fit.