jbruce12000/kiln-controller #225: Error: short circuit to ground only at higher temperatures

TinMaan, 2025-07-23:
Almost there!  After an auto-tune the result of calculations were entered in config file, ran the  " test 200-250 " almost perfectly. Next was a Bisque Fire test. I volunteered a few pots to be Bisque fired. Entered a custom profile and hit START. All went well all the way up to about 1100'F. It stopped the profile ( Run complete ) a couple of times. The third time it fired well to about 1600'F . I was getting " Error - Problem reading temp, short circuit " repeatedly. Shortly after, it announced " Run complete "  while the Pi's screen was covered with " Problem reading temp, short circuit "
I understand that the thermocouple and it's cable must be well insulated from circuit ground. I am sure, I have done all needed to be done about that issue. 
Why the system started acting up in higher temperatures?  And what is causing the " short circuit "?

rondoc, 2025-07-23:
This seems to be an issue for several users. I think that there are several
possible causes.
1. Thermocouple shorting to itself or to ground.
2. The pi board uses pins to make the connections . These are not a very
reliable connection method. I don't know a better way but the quality of
the pin wire combos varies. I found after some intermittent problem that
using new pins with good wire stopped the problem.
3. RFI the pi board is susceptible to RFI  from.any source. In kiln areas
it could be any motor that has brushes, relays, electronics of any type .
Very hard to pin down. One solution is to put a capacitor across the
thermocouple leads. 0.1 uF ceramic works fine.
4. A problem within  the pi  board itself. They are made by many sources.
Cheap enough to replace.
5. The code is being misread due to code problems . Try reinstalling it
might help
6. The.power source for the pi might be electrically noisy . Try a
different one.
You may never resolve the uncertainty so plan for it. I have not had these
problems and use my phone to watch the firing. Usually I am asleep during
the firing so wouldn't know what happened until opening time. I have
considered making a independent monitor but so far after hundreds of firing
no failures due to the pi system.
I still think an Arduino based system would be more reliable but not don't
it yet..may be later this year and I will release it as freeware.

TinMaan, 2025-07-23:
As you mentioned a few of the points you made, have been discussed here as possible causes for the thermocouple related errors. There are also a few interesting new possibilities you are sharing. I have to look into those now.
That said, it is not clear to me and in my case,  - why - these errors start to appear in higher ( above 1100'F ) temperatures?
Most if not all mentioned hardware and software issues if present, should be present throughout the firing.
I like to put the blame on the cheap thermocouple. Perhaps in higher temperatures, the hot-juction of this cheap unit starts to separate so ever slightly and send " irregular " signal that is read as " noise " by the system.
I will replace it with a new ( still cheap ) one and report back.
BTW, could you recommend a reliable source for purchasing " good enough " thermocouple?

TinMaan, 2025-07-23:
I am answering my question here in case someone else is interested. I leaned that, the physical signal from a thermocouple ( cheap or otherwise ) is not constant in time by lab standards. So, the signal generated by the thermocouples ( cheap or otherwise ) physically are random to the point of being considered badly noisy to begin with.  At lower temperatures the random behaviour proportional to temperature is handled by system as if they don't exist. At higher temperature the randomness of their signals increase proportional to temperatures they are exposed to - edging towards crashing the software if tolerances are set too tight. 
Problems begin when low quality thermocouples [ ADD ] more noise to the unavoidable random signal, especially at higher temperatures. It is silly to increase the acceptable tolerances for errors in the software to accommodate a cheap thermocouple. Therefore, I am buying a better thermocouple-set with cold-junction cover incase the kiln-controller will expand to include cold-junction compensation on day.

rondoc, 2025-07-23:
> learned that, the physical signal from a thermocouple ( cheap or otherwise

I don't think you've understood thermocouples, sorry. The thermocouple tc
is made by two dissimilar metals . They are welded at on end. The other end
goes to another   junction but this one is a virtual one as part of the
chip..the effect of this a very small voltage is produced by the
temperature difference. The voltage is millivolts so noise from any source
can be an issue. However if you connect the tc from.the kiln to the k type
converter board 55 type and the same  to a Fluke or other tc reader you
should see same temp on the Fluke as on the pi system. You need to.use tc
wires correctly oriented for this.
Please report and let us know how your system.is working.

TinMaan, 2025-07-23:
Yes, thank you for reminding the basic theory of the thermocouples. There are lot that I gradually started to understand about kiln building and controlling them thank to all contributors here and the internet. At this stage the practical problems and how to overcome them in execution of the ideas that concern me. 
My thermocouple is connected properly ( according to me ). It functions perfectly with consistent temperature control errors of amazing * 0.5 F' ~ 0.01 F' *  as long as readings are done bellow 1100'F. No overshoots no pauses for kiln to catch up - PERFECT by definition! ( thank you jBruce12000 ). But why the system crashes often above 1100'F!?

BlakeCLewis, 2025-07-23:
I think you may be misdirected by the error interpretation.
Looking at the MAX31856 datasheet, I do not see a fault classification that
would correlate to Shorted Thermocouple. I see an Open Thermocouple fault.
There are 6 different fault states reported by the max31856:
  Cold Junction out of range (over) (range is -55c to 128c)This is chip
temperature, thermocouple in the chip that assumes that the 'cold junction'
(junction where the thermocouple wires are connected to copper) is also at
this temp.
  Cold Junction out of range (under)
  Thermocouple out of range (over 1372C)
  Thermocouple out of range (under -250C)
  Input voltage either over/under (this is input power)
  Open Thermocouple (broken connection in thermocouple/wires)

I am thinking the software is interpreting the fault signal incorrectly.
The issue I have had was an Open Thermocouple. Heat cycles can
expand/contract connections and can cause loosing up connections. Also, had
problems with jumper wires as rondo said.
I have also had thermocouples not read until I tapped them. Like maybe a
corrosion issue between wire and thermocouple.
The connection between the pi and the MAX31856 has not been a problem for a
very long time. I have the modules mounted on a perma-proto pcb that is
made to plug into a Raspberry pi. Adafruit.com makes them. They are
laid out, and look like a breadboard, but are a pcb with a 40 pin connector
on it. "https://www.adafruit.com/product/2310". I put a chip on it that
converts pi signals to 12v for the relay coils. I also put the Max31856
module on it. Originally, I had made up crimped jumper cables. I had some
issues like you are having and converted to solder connection where
reasonably possible. I now do not have any unsoldered jumper wires. I also
had the thermocouple wires not engaged into the screw terminals on the MAX
correctly. I think it would loosen up and I would get intermittent bad
readings(downspikes). Like 0 C and the kiln would start running away. All
my issues have been the workmanship of connections.
Blake

TinMaan, 2025-07-23:
@BlakeCLewis, as they say, Like minds think alike. I had just finished replacing all possible contact points to soldered joints before checking here for replies. I will build another Pi board + protoboard you linked also for comparison and plan-B. 
I agree with you also on the possibility of error miss - interpretation by the software. After all the logic set is small and at growing stages.
I'll comb through all the possible failure points armed with the information I received here and tighten the knots again with the hope of making the system fail proof ( if that's practically possible ).

BlakeCLewis, 2025-07-23:
I had one failure recently: The kiln is on wheels and I rolled it a bit too
far away from the controller. The thermocouple wire is the shortest.
Another failure I had years ago was heat on the pi, so I added a fan to the
enclosure and switch it on if the cold junction temp is higher than the
room temp. I use a thermocouple(second max31856) for room temp. A
thermistor would have been cheaper.

rondoc, 2025-07-23:
Try and keep it simple. The 55 chip board is fine for a K type tc. The 56
is more versatile but unnecessary. I've used the 55 chip/board for at least
years. If you see in the setup of the pi the tc type is a parameter.
It's a PITA when simple stuff doesn't work. It's often quicker to just make
second system. Life is short use.your skills where they work.
Ronald.

chipgarner, 2025-07-23:
This hasppens consitently above about 800c. it continues when the power off on coolling

When the power is off to THE SSR's eg.  the kiln sitter trips, the error stops.

Same behavior on a kiln with mechanical relays

keebie81, 2025-07-26:
Ive had similar issues with my pi at higher temps. The temperature would flat line and not change, sometimes it would continually drop when the temperature was actually rising because kiln was full on. If I unplugged thermocouple from pi and directly in to my fluke meter it read perfectly fine at these high temps. So im assuming the fluke meter ignores errors or has better shielding. 

Running a ground from pi to the wall outlet ground helped it read at higher temps but the issue was still occurring near top range. I now instead have a ground wire from wall outlet running to one of the thermocouple connectors at block coming out of kiln.  

This has resolved the loss of reading at high temps for me. Im using a max31856.

rondoc, 2025-07-26:
This is first useful news on this issue. My kiln has a steel case that is
grounded . So may be thats why I dont have problems.
Can you undo your grounding and see if problem reoccurs?
Putting a capacitor across the tc wires right at the input module might
work to if it's an RFI  problem. My question is if it's RFI where is it
coming from?
BTW you can connect the Fluke up  to the same tc as the module. The input
on the Fluke will not affect the pi system.

keebie81, 2025-07-26:
[image]
[image]

> I just reread your post. Did you connect a tc wire to ground? This should
> not work!!
> The input to the 55 or 56 module is designed as a floating input. So no
> ground. Explain ground wire to thermocouple connector please.
> […](#)

Yeah i thought so too but it works for me. Unless my max31856 board has something weird going on. As seen in pic i have a wire going from positive terminal to ground of wall plug. My kiln outside is grounded also. I measure around 0.4-0.6ohm from case to ground of wall plug. Same from either connection of thermocouple block and kiln case.

rondoc, 2025-07-26:
From the spec sheet on the chip you are shorting out part of the tc. Unless
you are not connecting any other part of the pi system to ground. In other
words one side of the tc is grounded but still floating wrt the pi . That
would make sense but strange nevertheless.

TinMaan, 2025-07-26:
Could some one explain how the code " assumes " there is a short circuit present to begin with. What triggers that Error ?
I have been using these inexpensive thermocouples with Skutt and CRESS Kilns. Bartlett controllers crash now and then as result of over heating or brownouts but they never complain about short circuit somewhere in the system.

BlakeCLewis, 2025-07-27:
I was assuming, not the code.
It looks like the MAX31855, throws errors for short to Ground, and short to
Vcc. I looked over the project code and it combines these 2 fault states to
1 message "short circuit".
It would be nice to know if it is Vcc or Ground.
I think the max sends a pulse through the thermocouple and then looks at
the other end of the circuit to detect Open.
Short to Vcc: It measures the voltage compared to Vcc
Short to grnd: it messures the voltage compared to gnd

"
https://www.analog.com/media/en/technical-documentation/data-sheets/max31855.pdf
"
Table 3 has a description of the faults the MAX31855 reports.
It is up to the driver to put a label on the reported error state.
So, I was assuming that the error message matches the reported state.
and I was also looking at the MAX31856 datasheet, and the fault section is
different.