jbruce12000/kiln-controller #229: Circuit to ground error

rock1148, 2025-07-27:
I have triple checked the wiring and there is nothing touching a ground. I'm using a max31855 and have tried 2 different thermocouples (K type). I can bypass the error by setting the ignore_tc_short_errors to True in the config file, but I suspect I'll pay for it down the road.

Any troubleshooting suggestions would be helpful.

BlakeCLewis, 2025-07-27:
send pictures of tc, pi, ssr, and max wiring? Got to get it into my mind.

rondoc, 2025-07-27:
> Under what conditions do get short to ground errors.?

If it is at high temperatures it may be itscnot the temperature but the
length of time. Try making a temp profile that is slightly above ambient
with a run time if 4 hours.  See if it does this without errors. At thus
juncture there are too.many parameters to narrow down the cause. I don't
get  problem but several have reported it. What is the difference?

rock1148, 2025-07-27:
I have to set the ignore short off in order to make the program work. The firing goes up to 700c with no problem. After that the thermocouple reading lags behind the actual temperature and at 1150, the thermocouple reads about 950.

rondoc, 2025-07-28:
Do you mean that regardless of profile or temp.the program won't run unless
you delete the short tc part.
R.

Anger is usually regrettable, kindness rarely.

rock1148, 2025-07-28:
That’s correct, the program stops with the error short to ground.

rock1148, 2025-07-28:
Clarification: I can run the software using the old version (before this code was implemented):

> import board
>     spi_sclk  = board.D17    #spi clock
>     spi_miso  = board.D27    #spi Microcomputer In Serial Out
>     spi_cs    = board.D22    #spi Chip Select
>     spi_mosi  = board.D10    #spi Microcomputer Out Serial In (not connected) 
>     gpio_heat = board.D23    #output that controls relay
>     gpio_heat_invert = False #invert the output state

When running the most recent version - even if I set the "ignore_tc_short_errors = False" to True, I still get the short to ground error and the software simply loops forever repeating the error.

However, using the old version, it will run if I set the ignore_tc_short_errors = True.

rondoc, 2025-07-30:
> I used the kiln for over a year. I just changed the chip from 55 to 56 and
> no errors, is working perfectly. I suspect the 55 went "bad". Anyway, the
> problem is resolved. Thanks to all for your help.

It w asnt the change from 55 to 56 it was a bad board. For a K type the 55
board is the correct one .
Glad it's all working .
R

Anger is usually regrettable, kindness rarely.

rock1148, 2025-07-30:
I agree that it was a bad board, but I find it strange that it worked for
just over a year with no issues and then failed as I explained. I'll leave
the 56 board and for now use the K thermocouple, but I'll get one of the S
types from Amazon. I know they come from China and are cheap - hopefully
they are actually made with actual platinum / rhodium and not some strange
alloys. The difference in cost is amazing - Amazon $32, Pottery place $450!

rondoc, 2025-07-30:
The S type are very fragile as thin wire. The k type are very robust.

Anger is usually regrettable, kindness rarely.