jbruce12000/kiln-controller #231: Kiln run completed, even though it is not

Nikodemuzz, 2025-08-02:
Hi all!

I cannot seem to quite get my kiln controller to work properly. All the tests seem OK, but when running an actual kiln run it gives me the "Kiln run completed" although the program is not nearly there. At first try, this happened after less than half an hour. I then double checked all my wiring to the RPi and on the second try it already seemed like I had solved the problem. However, this time three hours into the program, the same thing happened (around 220C). If I try to restart the profile, it immediately stops and thinks it has completed.

This is the log file it created after the second try:

[kiln.logs.gz](https://github.com/user-attachments/files/21559127/kiln.logs.gz)

Nikodemuzz, 2025-08-02:
An update:

I managed to get the program to continue when I put the ignore_too_many_tc_errors to True. However, something does seem to be off with the thermocouple reading. After the program had been continuig for a while the temperature seemed suddenly to jump 200+ degrees after which it started to gradually cool. This screenshot is perhaps half an hour after the spike:

[image]

I'm pretty sure a jump like that is not actual, but a measurement error of some kind.

rondoc, 2025-08-04:
> Another update:
>
> I didn't touch anything after the temperature spike. It took an hour or so
> for the temperature reading to normalize, after which the controller
> resumed the program. There were no further issues and the run completed as
> expected. Unfortunately I didn't have a secondary temperature measurement
> to check if the reading within the controller was correct. Judging by a
> faint glow appearing at around 550C and the glow being very bright orange
> at peak temperature, it at least wasn't completely false.
>
> I don't really know what conclusions to draw from this. I'm happy that the
> run was completed in the end, but something is still a bit amiss I think.
> image.png (view on web)

Get a good tc reader and connect it in.parallel with the pi. It will give
you a check. I like the Fluke it has been reliable for about 40 years!!
Ronald

Anger is usually regrettable, kindness rarely.

Nikodemuzz, 2025-08-04:
I agree that a parallel measurement is a good idea. In my professional life, hooking up a single TC to more than one device has been a no no. Installing a secondary TC is not a big deal though, I might do that. I have another kiln run underway, a glaze firing this time. I have some Orton cones in there to give an idea where I'm at, although they don't really measure temperature specifically. It will be interesting to see if I get more of these temperature spikes this time.

rondoc, 2025-08-05:
> What check ( information !? ) would a thermocouple reader provides in
> addition to the GUI, the terminal window and the log? Can I use an average
> Pyrometer instead of a Fluke Multimeter?

It's a check on the tc and should be same temp.within a few degrees of that
via the pi system .
Ronald.

Anger is usually regrettable, kindness rarely.

stefan63, 2025-08-17:
The "Kiln run completed" issue has occurred a few times for me as well (Using a PI Zero W V1.0) and It seams like I need to restart the Pi from time to time to get a stable comunication (Wifi Issue?).
However this is a superr usefull tool and I mainly use it for anealing 3D prints, paint and epoxy work and drying 3D printer filaments.
I use a cheep mini oven with built in fan in the sub 100 euro segment .
REF:https://www.stockholmviews.com/wp/3d-print-annealing-of-pla-and-ht-pla/