jbruce12000/kiln-controller #232: PIP install errors on new controller build

sjohnsongitstuff, 2025-08-03:
I'm running into a problem during the initial install of the controller.  I'm using a Pi ZeroW with the latest Raspberry Pi Imager.  The install goes fine until the "pip install" portion, which errors out on the greenlet package.  I don't know much about python so I'm not sure where to look next.  The error text is listed below.

(venv) pi@Kiln:~/kiln-controller $ pip install greenlet
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting greenlet
  Downloading greenlet-3.2.3.tar.gz (185 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: greenlet
  Building wheel for greenlet (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for greenlet (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [117 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build/lib.linux-armv6l-cpython-311/greenlet
      copying src/greenlet/__init__.py -> build/lib.linux-armv6l-cpython-311/greenlet
      creating build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/test_generator.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/test_weakref.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/test_gc.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/fail_switch_three_greenlets.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/fail_switch_two_greenlets.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/fail_cpp_exception.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/test_greenlet_trash.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/__init__.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/test_contextvars.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/test_extension_interface.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/test_greenlet.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/leakcheck.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/test_cpp.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/test_version.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/test_tracing.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/fail_initialstub_already_started.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/test_throw.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/fail_clearing_run_switches.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/test_stack_saved.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/test_leaks.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/fail_slp_switch.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/test_generator_nested.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      copying src/greenlet/tests/fail_switch_three_greenlets2.py -> build/lib.linux-armv6l-cpython-311/greenlet/tests
      creating build/lib.linux-armv6l-cpython-311/greenlet/platform
      copying src/greenlet/platform/__init__.py -> build/lib.linux-armv6l-cpython-311/greenlet/platform
      running egg_info
      writing src/greenlet.egg-info/PKG-INFO
      writing dependency_links to src/greenlet.egg-info/dependency_links.txt
      writing requirements to src/greenlet.egg-info/requires.txt
      writing top-level names to src/greenlet.egg-info/top_level.txt
      reading manifest file 'src/greenlet.egg-info/SOURCES.txt'
      reading manifest template 'MANIFEST.in'
      warning: no previously-included files found matching 'benchmarks/*.json'
      no previously-included directories found matching 'docs/_build'
      warning: no files found matching '*.py' under directory 'appveyor'
      warning: no previously-included files matching '*.pyc' found anywhere in distribution
      warning: no previously-included files matching '*.pyd' found anywhere in distribution
      warning: no previously-included files matching '*.so' found anywhere in distribution
      warning: no previously-included files matching '.coverage' found anywhere in distribution
      adding license file 'LICENSE'
      adding license file 'LICENSE.PSF'
      writing manifest file 'src/greenlet.egg-info/SOURCES.txt'
      copying src/greenlet/CObjects.cpp -> build/lib.linux-armv6l-cpython-311/greenlet
      copying src/greenlet/PyGreenlet.cpp -> build/lib.linux-armv6l-cpython-311/greenlet
      copying src/greenlet/PyGreenlet.hpp -> build/lib.linux-armv6l-cpython-311/greenlet
      copying src/greenlet/PyGreenletUnswitchable.cpp -> build/lib.linux-armv6l-cpython-311/greenlet
      copying src/greenlet/PyModule.cpp -> build/lib.linux-armv6l-cpython-311/greenlet
      copying src/greenlet/TBrokenGreenlet.cpp -> build/lib.linux-armv6l-cpython-311/greenlet
[... 48 lines cut ...]
      copying src/greenlet/platform/switch_x32_unix.h -> build/lib.linux-armv6l-cpython-311/greenlet/platform
      copying src/greenlet/platform/switch_x64_masm.asm -> build/lib.linux-armv6l-cpython-311/greenlet/platform
      copying src/greenlet/platform/switch_x64_masm.obj -> build/lib.linux-armv6l-cpython-311/greenlet/platform
      copying src/greenlet/platform/switch_x64_msvc.h -> build/lib.linux-armv6l-cpython-311/greenlet/platform
      copying src/greenlet/platform/switch_x86_msvc.h -> build/lib.linux-armv6l-cpython-311/greenlet/platform
      copying src/greenlet/platform/switch_x86_unix.h -> build/lib.linux-armv6l-cpython-311/greenlet/platform
      running build_ext
      building 'greenlet._greenlet' extension
      creating build/temp.linux-armv6l-cpython-311/src/greenlet
      arm-linux-gnueabihf-g++ -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/home/pi/kiln-controller/venv/include -I/usr/include/python3.11 -c src/greenlet/greenlet.cpp -o build/temp.linux-armv6l-cpython-311/src/greenlet/greenlet.o
      src/greenlet/greenlet.cpp:16:10: fatal error: Python.h: No such file or directory
         16 | #include <Python.h>
            |          ^~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/arm-linux-gnueabihf-g++' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for greenlet
Failed to build greenlet
error: failed-wheel-build-for-install

× Failed to build installable wheels for some pyproject.toml based projects
╰─> greenlet
(venv) pi@Kiln:~/kiln-controller $

sjohnsongitstuff, 2025-08-03:
Update to the problem:

I found the issue.  The current Raspberry Pi Imager ( v1.9.6) does not include the Python3-Dev package in the OS i am using (Raspberry Pi OS Lite 32-bit).  I installed the package and everything downloaded and compiled without a problem.

rondoc, 2025-08-04:
If you copy the code to chatgpt it will give you some answers. Worth a try.

Anger is usually regrettable, kindness rarely.

snaeil, 2026-10-02:
This should be at least documented IMO .
Could you share how you solved this @sjohnsongitstuff ? 
Did you install pip system wide before?
It would be nice to have a reproducible list of commands in the documentation