jbruce12000/kiln-controller #241: Is the system reading temperature " while " heating?

TinMaan, 2025-08-07:
Could a code expert clarify this subject please? In another forum, I encountered a few variations of errors caused by max31855 reading temperature " WHILE " controller is heating rapidly. Although there are quite a few, the one that most interest me at this point is : Possibilities of Error report by controller if, ... we are reading the temperature while heating causes too many changes in temperature in one cycle to the point the system flags the behaviour. 

Does Kiln-Controller read temperature while it heats the kiln!? If yes, could that be a source of errors in higher temperatures!?

rondoc, 2025-08-08:
Kiln temp.rise is far too slow to affect the reading. Find another
culprit...

Anger is usually regrettable, kindness rarely.

TinMaan, 2025-08-08:
> Kiln temp.rise is far too slow to affect the reading. Find another
> culprit...
> 
> Anger is usually regrettable, kindness rarely.
> […](#)
I am not quite sure what constitute slow in this context. Would you share some numbers please? My question is on the subject of synchronous " reading / heating " effects especially in higher temperatures. The max31855 subroutines for fault detection are complex and unforgiving. Because of these subroutines some heating systems make sure the " READ " cycle by 55 board starts and ends after each " heat " cycle and never over lap.

rondoc, 2025-08-08:
You bring up some interesting points . I will.try to study the code for
reading temp  and see if there is a timing issue. Someone far more skilled
than I can.probably answer or may be AI can solve the problem.
Two things worth doing 1. Add a 100 ohm resistors in each tc lead and 10NF
cap across the actual board. A third thing is put a 1uF cap across the vcc
on the breakout board.
Ronald.

Anger is usually regrettable, kindness rarely.